ArduPilot/ardupilot #16163: Assign unique SYSTEMID to SITL and QGC.

muramura, 2020-12-29:
## Feature request
I use SITL, MISSION PLANNER, QGROUNDCONTROL, and APM PLANNER2 to debug.
I have a communication anomaly with FC due to SITL's GCS, MISSION PLANNER, and QGROUNDCONROL using the same system ID.
I, APM PLANNER2, do not occur because of 252.
I think it is better to assign unique ID.

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
[image]

khancyr, 2020-12-29:
there is already unique identifier : 
https://mavlink.io/en/messages/common.html#MAV_COMPONENT for the routing and https://mavlink.io/en/messages/common.html#MAV_TYPE on the heatbeat

muramura, 2020-12-29:
@khancyr san.
I am not able to set the component ID in the MISSION PLANNER, QGROUNDCONTROL settings.
SYSTEM ID is used for the following connections.
I think it is better to assign a unique ID to major GCS.
https://ardupilot.org/dev/docs/mavlink-routing-in-ardupilot.html

MISSION PLANNER
[image]

QGROUNDCONTROL
[image]

IamPete1, 2021-08-16:
I'm not really sure what the action we should take on this, so I'm going to close, @muramura please re open if there is some code change in AP that would revolve.

closed 2021-08-16 by IamPete1