ArduPilot/ardupilot #16177: Quadplane/Copter: Allow pilot nudging of position in AUTO modes

cheintz, 2020-12-30:
## Feature request

**Is your feature request related to a problem? Please describe.**

See issue #16069

When auto-landing a quadplane using a Q_LAND waypoint, the elevator and ailerons respond to stick commands (pitch and roll, which correspond roughly to forward and lateral velocity in position controlled modes), but the quad motors do not. The desired pitch and roll do change in response to the stick commands, but these desired values are not achieved. Yaw control does use the quad motors and appears to track the commanded value.

This is especially concerning in a VTOL/Quadplane, configured with STICK_MIXING = 1, where the pilot has an expectation of being able to adjust the trajectory in auto mode without leaving auto mode, but does not have this ability if the aircraft is flying in a VTOL mode.

**Describe the solution you'd like**
A good solution would be a parameter like the Plane parameter STICK_MIXING, with behavior similar to 1=FWBA Mixing, where the pilot's stick inputs influence the trajectory, but releasing the sticks restores normal auto mode. Perhaps behavior in following a NAV_VTOL_LAND, the sticks influence the desired landing position (as in the Q_LAND mode) and releasing the sticks after applying the sticks results in a straight-down descent, rather than a return to the original landing waypoint.

**Describe alternatives you've considered**
The only alternative as of now is to switch into Q_LOITER, make the adjustment, and switch back into AUTO. For augmented land, the pilot could switch into Q_LAND to have the decent rate managed but influence the landing position with the sticks. On a Quadplane, this is getting to be a lot of modes mapped to the TX. 

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[X ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
Logs (same as linked closed bug report):

[2020-12-10 14-10-05.zip](https://github.com/ArduPilot/ardupilot/files/5698644/2020-12-10.14-10-05.zip)

rmackay9, 2020-12-30:
For copter there are several similar issues in the list:

- https://github.com/ArduPilot/ardupilot/issues/11251
- https://github.com/ArduPilot/ardupilot/issues/7649
- https://github.com/ArduPilot/ardupilot/issues/6796
- https://github.com/ArduPilot/ardupilot/issues/6391

robustini, 2021-01-02:
Stick mixing would certainly be an excellent feature, even if theoretically during any mission if well programmed there should be no need to give roll or pitch commands, while the yaw does.