ArduPilot/ardupilot #16178: Documentation update needed

1AwesomeDeveloper, 2020-12-31:
## Feature request

**Is your feature request related to a problem? Please describe.**
I am new to ardupilot, so I was going through the documentation and on his page https://ardupilot.org/dev/docs/apmcopter-programming-libraries.html# , I found that the documentation didn't mention all the libraries that are in the code
**Describe the solution you'd like**
PLEASE UPDATE THE DOCUMENTATION, IF POSSIBLE PLEASE ADD  A BRIEF OF ALL THE REMAINING LIBRARIES

magicrub, 2020-12-31:
For being an awesome developer, you should have made a pull request instead of an issue! :)

Take a look at this video
https://www.youtube.com/watch?v=fb73F8_MiMg

1AwesomeDeveloper, 2020-12-31:
HaHa, thank you, sir, will try to do it now but for that, I guess I need to know about all these libraries and that is why I was asking for documentation, 
Nevertheless, I will try to study code and see if I can submit a PR for the same 

Thank you

IamPete1, 2021-07-18:
Closing, should be a issue over on the wiki repo, https://github.com/ArduPilot/ardupilot_wiki

closed 2021-07-18 by IamPete1