ArduPilot/ardupilot #16193: Rover: Barometric pressure sensor number 0 is not enabled.

muramura, 2021-01-01:
## Bug report

Altitude information is not required for the rover to run.
I am not using the WP altitude information.
I set the number of barometric sensors to 0.
FC notifies me "Baro: unable to initialise driver".
I think rover should enable barometric sensor number 0 as well.

**Issue details**

_Please describe the problem_

**Version**
ArduRover V4.0.0 (0e52bafa) 

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

**Airframe type**

**Hardware type**
SITL

**Logs**

[image]

IamPete1, 2021-08-16:
I think @andyp1per has been playing with flight controllers that have no baro. Maybe there is some param config that will allow it.

andyp1per, 2021-08-16:
You need to define 
```
define HAL_BARO_ALLOW_INIT_NO_BARO
```
in hwdef.dat