ArduPilot/ardupilot #16202: Matek H743 Soft Brick

Almischalmi, 2021-01-02:
## Bug report

**Issue details**

Matek H743 soft bricks after a while. It seems to be stuck in startup mode. After reflashing it with STM32CubeProgrammer it works again.
I am not the only user experiencing the issue. 
See also: 
https://www.rcgroups.com/forums/showpost.php?p=46112407&postcount=195 ff.
and
https://discuss.ardupilot.org/t/watchdog-reset-in-flight-and-ardupilot-does-not-complete-startup-and-hangs/64447

**Version**
4.0.7 and 4.1.0-Dev

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Flying Wing

**Hardware type**
Matek H743

**Logs**
Haven't been able to record logs when it happens.

magicrub, 2021-01-03:
What commit were you on?

Almischalmi, 2021-01-03:
Must be the one after 1c57eed66caf259b2bb6ab42ba6df8f0ce363f2a.

philippbenkert, 2021-01-03:
Same prioblem with me. With different versions from 4.07-4.10.
After a few successful flights, the FC no longer boots. It is also no longer found on the PC via USB.
I flew a round with the Dodecahexa today. Then I wanted to set 4 waypoints and the next test nothing worked.

yaapu, 2021-01-13:
probbaly related to this PR https://github.com/ArduPilot/ardupilot/pull/16304

andyp1per, 2021-01-13:
FWIW I am testing one of these boards and there is significant DMA contention - I can see that this would cause issues for things involving DMA

tridge, 2021-01-15:
> probably related to this PR #16304

i don't think it is related

yaapu, 2021-01-15:
> > probably related to this PR #16304
> 
> i don't think it is related

ok thanks tridge

stevereed66, 2021-02-06:
This is the post I used https://discuss.ardupilot.org/t/watchdog-reset-in-flight-and-ardupilot-does-not-complete-startup-and-hangs/64447

I am having same problem
can reproduce as follows.
arduplane 4.08
matek  h443 wing
quad plane y3 2 tilt rotors
No logs -- will try log before arming.

Task calibrate current sensor
connect ground station 3dr telemetry.
reverse props.
qstabilize.
arm
power half throttle say 20A measures with wht meter.
edit current measured to 20A. when saved motors stop, no response from  APM.
Reboot normal LED then  permeant red and greenled permanent buzzer, no USB detected. Reading code seems to mean flash memory corrupt.
re-flash arduplane.apj 4.08 still the same.
Flash  MatekH743_bl.hex indfu mode with inave  configurator (noerase), Flash arduplane.apj 4.08 still the same.
Fix
Flash  arducopter or any firmware other than plane. Flash arduplane   all working again.

Also if you follow the issues link and run  zero.bin  run  MatekH743_bl.hex  flash arduplane also fixes.

I have not seen any posts this cause crashes, is this the case. Seems intermittent when saving params.

Also I would like to try 4.1 and the CRSF protcol, where can I get latest status of 4.1 Dev

Thanks

Steve

andyp1per, 2021-02-06:
@stevereed66 can you try the same procedure on 4.1? There have been some significant fixes on 4.1 which I suspect resolves this for you. You can find 4.1-dev here: https://firmware.ardupilot.org/Plane/latest/

stevereed66, 2021-02-06:
Yes, Is  4.1 at a stage to fly, would like to try CRSF  and RCIN

Get back tomorrow midday gmt.

andyp1per, 2021-02-06:
I have been flying it without issue (inc CRSF) - but I don't think your procedure require flying?

stevereed66, 2021-02-07:
Ok loaded 4.1
commit 164da261f10f0f6559330cf1d2dba875dd757eea
Author: Peter Barker <pbarker@barker.dropbear.id.au>
Date:   Sun Feb 7 08:20:57 2021 +1100
did same test. saved measured current 20 times. All worked as expected.

While I have your attention, hope you dont mind
Set up CRSF on serial 3 protocol 23 rcin baud 57
all working telemetry, RSSI and RC control but the FM opentx sensor has now appeared but only shows QSTB mode and does not change as mode changes. 

You are doing a great job, stay safe

closed 2021-03-17 by Almischalmi

zhaodelong, 2021-04-07:
Same prioblem with me. With different versions 4.09 stable.
After a few successful flights, the FC no longer boots. It is also no longer found on the PC via USB. Has to burn the firmware again. This is the third time this happens.

rojulec, 2021-05-25:
Same problem for me. Version 4.09 stable.
About three flights and the H743 now hangs on boot and no longer on USB port.
Inital failure happened in flight just after take off. Motor disarmed and I safely landed while I had control.
After repowering FC, it would not successfully boot up.
I have a log (bin file) saved in sd card apm log folder.
It it worth trying to work out what happend there or just wait for 4.10 stable to be released?
[00000021.zip](https://github.com/ArduPilot/ardupilot/files/6537081/00000021.zip)