ArduPilot/ardupilot #16231: PreArm: Internal errors (0x3000)

muntasir399, 2021-01-05:
[image]

muntasir399, 2021-01-05:
> Which hardware?

pixhawk 2.4.8

rmackay9, 2021-01-06:
we actually have this issue now to collect up all the internal error reports: https://github.com/ArduPilot/ardupilot/issues/15916

hendjoshsr71, 2021-09-21:
Should this one be closed? There is no log or info. And nothing got put into the Internal reporting page.

@muntasir399  Have you had this issue on recent versions of the software? Say Copter 4.1 betas or release candidates for example?

IamPete1, 2021-10-01:
@muntasir399 if this is still a issue on 4.1, please re-open with some more information, ideally a log and param dump.

closed 2021-10-01 by IamPete1

ZhongmouLi, 2023-12-17:
Hi guys, I believe this happens also on Copter 4.4.0 and my hardware is KakuteH7.

amilcarlucas, 2023-12-18:
@ZhongmouLi have you looked at #15916 ? you should.