ArduPilot/ardupilot #16235: Tradheli: replace copter specific frame config define with universal frame config define

bnsgeyer, 2021-01-06:
**Issue Details**
A universal frame config define (libraries and greater code) for tradheli called 'HELI_BUILD' was recently merged.  The older frame config define called 'HELI_FRAME' that only applied to the greater code needs to be replaced by the universal frame config define.

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Heli_single, Heli_Dual and Heli_quad

**Hardware type**
N/A

**Logs**
N/A

IamPete1, 2021-09-06:
@bnsgeyer any more info? I cant find a `HELI_BUILD` define.

bnsgeyer, 2021-09-06:
@IamPete1 i ran into an issue with heli autotune where I wanted to define a set of #defines based on the heli build. Similar to this line
#if APM_BUILD_TYPE(APM_BUILD_ArduPlane)
in the current autotune.cpp file.  Currently heli specific build items are only broken out in the greater code with the frame_config statement which can’t be used in the libraries to specify heli unique code. 
@tridge asked me to put this in as an issue in hopes that someone might help out.

IamPete1, 2021-09-06:
Looks like we don't have such a option for `APM_BUILD_TYPE` yet.

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Vehicle/AP_Vehicle_Type.h#L22-L33

I guess step one is to add that, then replace the existing uses `HELI_FRAME`, will probbibly have to change lots of `APM_BUILD_TYPE(APM_BUILD_ArduCopter)` to `APM_BUILD_TYPE(APM_BUILD_ArduCopter) || APM_BUILD_TYPE(APM_BUILD_ArduHeli) `

bnsgeyer, 2021-09-06:
@IamPete1 any chance you or @Gone4Dirt  would like to take on this little project. It would be very helpful for the Heli auto tune which I will be in the process of cleaning up and preparing for a master in the coming weeks.

Gone4Dirt, 2021-09-06:
Yeah, no problem @bnsgeyer I will sort this out.  Will likely be next week before I can look at it but should be able to get it done in plenty of time for you.

bnsgeyer, 2021-09-06:
Awesome. Thanks @Gone4Dirt

closed 2021-09-29 by Gone4Dirt