ArduPilot/ardupilot #16248: Plane: reach the target altitude of the next waypoint immediately during AUTO, respecting the limits of the TECS

robustini, 2021-01-06:
With Ardupilot while performing a waypoint flight, the plane flies along a virtual straight line between the previous and the next waypoint.
In the aeronautical world a new flight level is usually executed immediately.
It would therefore be interesting in Ardupilot there was an option to do the same thing.

- altitude of the waypoint just reached: 10 meters
- altitude of the waypoint to reach: 110 meters
- distance between the two waypoints: 1000 meters
- cruise nav speed: 20 m / s
- maximum height change set in the TECS: 10 meters per second

Currently the plane would be 500 meters away (halfway) between the two waypoints after 25 seconds at a height of 60 meters.
Instead what it should do with the new feature is to reach an altitude of 110 meters only after 10 seconds, at a distance of 200 meters from the first waypoint.
Same thing will do if the next waypoint is at a lower height.
This is aeronautically more logical, the manned planes behave this way during a flight for example in IFR.
And of course this should be an option that can be activated with a parameter, perhaps that you can choose whether this behavior occurs only during a climb, a descent or both.
This solution could also optimize consumption and it might also be interesting for the Copter.
I ping @tridge and @magicrub because they sure could be involved in this.

dcamatti, 2021-01-06:
@robustini , I fully agree with your request. I I want an altitude change, especially just after takeoff, I want to get rid of obstacles as soon as possibile using the max climb performance of the UAV. Like it is today I can use the max climb performance only by manually placing an additional wpt close enought to cause the UAV climb at max climb rate. Looking forward to have this implemented. Thanks - Daniele Camatti

peterbarker, 2021-01-06:
On Wed, 6 Jan 2021, Marco Robustini wrote:

> Instead what it should do with the new feature is to reach an altitude of 110 meters only after 10 seconds, at a distance of 200 meters from the first
> waypoint.

I do like the idea.

I have concerns about changing the way the vehicle executes missions based 
on parameters.

Better to represent this in the mission IMO, in a standardised fashion. 
That should allow you to move the mission to another flight stack and get 
the same results.

There is precedence in do-change-speed.

Command could also set a percentage of your maximum climb (or sink) 
rate to be used to reach the new altitude.

Would also need a way to cancel the behaviour.

robustini, 2021-01-07:
Hi @peterbarker! I really wanted to avoid this thing being implemented using further events, in complex missions you should use many, but as a last resort it can be fine.
In my opinion it would be more useful if this mode was selectable with a parameter, perhaps using an existing one that includes a bitmask.
Arduilot would also be less busy doing calculations, currently doing many more to maintain the line of flight between two waypoints at different heights.
My two cents...

AndKe, 2021-07-01:
I wish for a smooth altitude change between two points, arduplane is now just climbing to target altitude of the next WP before arriving there.
Is GLIDE_SLOPE_MIN still the way to make it distribute the altitude change across the distance between the two waypoints?

magicrub, 2021-07-01:
@AndKe did something change that I'm not aware of?

AndKe, 2021-07-01:
Hi @magicrub  not really, I am not aware of any sudden change.
We are flying fast and low over an airstrip as a part of a research project.
Between certain waypoints, like WP10 to 11,  the plane is supposed to closely follow the gradient of the airstrip. 
We observe that it quickly climbs to the elevation of WP12, right after passing WP10.  - instead of climbing gradually to it's altitude.
See center of this graph:

[image]

Log: https://drive.google.com/file/d/1Yo4zXbEi2yZXjywz2OkLQIav8RNsq_Ux/view?usp=sharing

WickedShell, 2021-07-01:
We normally build a slope and want to arrive at the target altitude when we
reach the waypoint. GLIDE_SLOPE_MIN is used to say when we are this many
meters over the target slope to the next point, and we are trying to climb,
recompute the slope. It's mostly there for long climbs, or right after
takeoff when we'd rather not push the nose back down to perfectly
reintercept the slope and waste energy.

AndKe, 2021-07-01:
@WickedShell   Thank you.
Please have in mind that this is not takeoff, just a low pass (coming from a higher alititude, flying low while following terrail altitudes, then climbing)
So the GLIDE_SLOPE_MIN requires 15m  to compute a glide slope, while the waypoints in question are at 10 and 14 meters (4 meters apart) 
What would be the downside of setting GLIDE_SLOPE_MIN to 3m ?
And why are we limiting the glide slope to changes greater than x meters ?

IamPete1, 2021-08-16:
Same idea as https://github.com/ArduPilot/ardupilot/issues/18157

lopsided98, 2023-01-13:
I believe the behavior @AndKe described is caused by this code: https://github.com/ArduPilot/ardupilot/blob/08f652fe2ca0bc9cb054b3557d8f58367712153b/ArduPlane/altitude.cpp#L63-L72

AndKe, 2023-01-13:
@lopsided98 good catch!  I would like to see it removed, or disabled by default. This behavior is not described anywhere and makes it harder to understand the system unless told about. 
Good system-knowledge is essential for safety.

shancock884, 2023-10-11:
Hi!
I've just come across this thread, as we had a similar question as @AndKe...
Ideally we would like to be able to use a glideslope when both ascending and descending, but from reading this, and looking at the code a bit, my understanding is that at present:
For descending:
- To always descend quickly, then fly flat, you can set GLIDE_SLOPE_MIN to 0 (or a big number).
- To descend on a glideslope as much as possible, you can set GLIDE_SLOPE_MIN to 1 (it looks to be an int16 param, so it would not be possible to set it less than 1m).
For ascending:
- Ascent will always ascent as quickly as possible, and then fly flat, and flying a climbing glideslope between waypoints is not supported.

Is my understanding correct? 
It would be nice if there could be a way to fulfil the wishes of both camps!

lopsided98, 2023-10-11:
I believe you are correct except for this point:

> * Ascent will always ascent as quickly as possible, and then fly flat, and flying a climbing glideslope between waypoints is not supported.

The glide slope logic works the same for ascent as it does for descent, unless the aircraft altitude is below 20 meters, in which case it will climb as quickly as possible.

shancock884, 2023-10-12:
Ah yes I see! Thanks a lot @lopsided98 - spot on! .... not sure why I got in a muddle reading that logic last night!!

eppravitra, 2024-07-07:
I just made a PR #27477 which should do what has been suggested here.

robustini, 2024-07-27:
> I just made a PR #27477 which should do what has been suggested here.

Thanks for the implementation!

closed 2024-07-27 by robustini