ArduPilot/ardupilot #16267: Feature Request:AP_Math:Quaternions:Improvements in the Quaternion functions

arshPratap, 2021-01-08:
## Feature request

**Is your feature request related to a problem? Please describe.**
While reading along with the Unreal Engine Docs, I came upon their Math library and in particular their documentation of [Quaternions
](https://docs.unrealengine.com/en-US/API/Runtime/Core/Math/FQuat/index.html) and after reading through their and Ardupilot's codebase on [Quaternion](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Math/quaternion.h), I felt Ardupilot was lacking in some functions.
**Describe the solution you'd like**
As stated above, I have made a comparison table between Unreal and Ardupilot's respective Quaternion class , which can be seen [here](https://docs.google.com/document/d/1OU-JkVmF0rp2r4kxpaFWi6mG81BkupmCR3kp_RtwlXc/edit?usp=sharing).It would be good if we can add some new functions where this class(Quaternion) feels lacking.
**Describe alternatives you've considered**
I felt that we can add some interpolation based functions as I have seen them being used in games.Some examples include First Person Escape Games,Turret and Tank Shooters and sometimes blending in different animations.

Thanks in Advance!!  
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

rmackay9, 2021-01-08:
We have a golden rule in AP that is we don't add code unless it is required so I think we need some more concrete ideas on how this would improve AP's ability to control a vehicle or add some other features users will appreciate.

I think this might close this because it's a bit too vague but happy to re-open if we come up with more concrete ideas.

closed 2021-01-08 by rmackay9

arshPratap, 2021-01-08:
Thank you, Sir

peterbarker, 2021-01-08:
@arshPratap Can you give us some examples where the "Could be added" functions could be used in the ArduPilot codebase?

We've got a "no dead code" rule, so we don't usually add things in on the off-chance they'll be useful in the future.