ArduPilot/ardupilot #16268: Issue With MavLinkFTP for QGC

kshitizgoyal5, 2021-01-08:
Hello, I am successfully doing FTP using PX4 and QGC. Now since my application is based on ardupilot stack, I need to use mavlink FTP with QGC but i am facing error as shown below.
[image]
The image above shows that I get an error of noAvailableSessions (errno5).
Kindly refer to the following for exact understanding https://github.com/mavlink/qgroundcontrol/issues/9306

liang-tang, 2021-01-14:
you can test it using missionplanner first.

IamPete1, 2021-08-16:
QGC does not support ArduPilots FTP scheme. I think the general consensus is that AP is not going to support two FTP protocols. Instead we should work with QGC maintainers so they can add support for the AP protocol. You could open a issue over on the QGC GitHub, if there is not one already.

closed 2021-08-16 by IamPete1