ArduPilot/ardupilot #16272: flightaxis SITL - incorrect current reading when negative values

nhadrian, 2021-01-08:
## Bug report

**Issue details**

Issue is detailed here:
https://github.com/ArduPilot/MissionPlanner/issues/2535

**Version**
4.1.0.0 (SITL)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Plane and quadplane

**Hardware type**
SITL

**Logs**
https://github.com/ArduPilot/MissionPlanner/issues/2535#issuecomment-756616999

IamPete1, 2021-01-13:
which realflight model and version of realflight?

nhadrian, 2021-01-13:
Realflight 9.5 and Griffin (the one I published). But I was able to reproduce it even with a custom quadrocopter in acro mode (where you suddenly pull down throttle in inverted position hence the negative current calculated by FR9.5).

IamPete1, 2021-01-16:
I can't reproduce this, as far as I can tell the voltage and current in the realflight NavGuides are just nonsense, for example:

[image]

Voltage clearly changes with throttle, If I were to guess its just the voltage and current of the first motor in the model. Its also possible to get the NavGuides to read steady state negative, for example:

[image]

So there cannot be a unit wrap issue, or we would see it there too.

So basically I don't think the value it that NavGuide should be used as a point of comparison and the AP value seems in the right ball park to me.

PR #16347 (2021-01-18): SITL: avoid negative current and voltage in flightaxis
RealFlight9 will sometimes give meaningless negative values
this should fix #16272
closed 2021-01-16 by IamPete1