ArduPilot/ardupilot #16278: APM: PreArm: GPS is not healthy: airsim and ardupiot SITL

djdylex, 2021-01-08:
## Bug report

**Issue details**

When i enter "python sim_vehicle.py -v ArduCopter -f airsim-copter --console --map" in the cygwin console in /tools/autotest, i am told "APM: PreArm: Need Position Estimate", 
"Got COMMAND_ACK: COMPONENT_ARM_DISARM: FAILED", and then
 "APM: PreArm: GPS is not healthy". 

The wierd thing is i could have sworn I had this working earlier, then all of a sudden it has broken.

**Version**
Master, latest commit.

**Platform**
[  ] Copter

**Airframe type**
Quad

**Hardware type**
SITL
Windows 10
Cygwin

**Logs**
_Please provide a link to any relevant logs that show the issue_

fabioandraderj, 2021-07-13:
Any idea about how to go around this?

IamPete1, 2021-08-16:
closing in favor of https://github.com/ArduPilot/ardupilot/issues/15329

closed 2021-08-16 by IamPete1