ArduPilot/ardupilot #16286: Ubuntu install script has incorrect test for realpath and libtool-bin

nabeelsherazi, 2021-01-09:
## Bug report

**Issue details**

The Ubuntu install script includes tests to see if realpath and libtool-bin are already installed. These tests are written incorrectly, as it is obvious to see:

```bash
# Check if we need to manually install realpath
RP=$(apt-cache search -n '^realpath$')
if [ -n "$RP" ]; then
    BASE_PKGS+=" realpath"
fi

# Check if we need to manually install libtool-bin
LBTBIN=$(apt-cache search -n '^libtool-bin')
if [ -n "$LBTBIN" ]; then
    SITL_PKGS+=" libtool-bin"
fi
```

The `-n` test [evaluates to true if the length of the string is nonzero](https://linux.die.net/man/1/test). Therefore, if e.g. realpath is found, `$RP` is a nonzero length string, `-n "$RP"` evaluates to true. In this case we would **not** want to try to install realpath again, the opposite of the current scripting. This is a common mistake in using `-n`.

This bug caused me an issue install on WSL2 where the script halts with error

```
Preparing to unpack .../realpath_1.19_amd64.deb ...
Unpacking realpath (1.19) ...
dpkg: error processing archive /var/cache/apt/archives/realpath_1.19_amd64.deb (--unpack):
 trying to overwrite '/usr/bin/realpath', which is also in package coreutils 8.28-1ubuntu1
Errors were encountered while processing:
 /var/cache/apt/archives/realpath_1.19_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

because the script is trying to install realpath when it's already present from coreutils.

The fix for this obviously easy and I'll submit it myself as a PR, but I wanted to submit an issue as well to reference.

**Version**
Latest as pulled from master 4d19ff6.

**Platform**
All

**Airframe type**
All

**Hardware type**
All

**Logs**
Above

khancyr, 2021-01-09:
Hello,

No we aren't checking if the package is installed, we check if it is available in the apt repo ! Therefore the check is correct.

What version of ubuntu are you using ?

nabeelsherazi, 2021-01-09:
Oh my godddd, my apologies. I really thought I had this figured out and I got it totally wrong. However -- there's definitely still something here because in the original version, my build fails as described above. I'm using 18.04 LTS Bionic on WSL2.

So the check, as written sees if realpath is an available package in the apt repo. If its available, it requests installation. This causes a problem on my machine because, as in the logs above, realpath is already present in all supported versions of Ubuntu (Trusty and up) via coreutils.

Could this possibly be fixed by checking for coreutils instead of simply seeing if realpath is available? So sorry for the brashness and impatience again! 😅

khancyr, 2021-01-09:
I will check again on a fresh VM, but normally the script is working correctly

khancyr, 2021-01-09:
I have checked on a fresh 18.04 and it is working as intend. Realpath isn't installed.
Did you have upgrade you machine from 16.04 to 18.04 ? 
I don't have WSL so I cannot test more.

nabeelsherazi, 2021-01-09:
Sounds good! I don't have a fresh machine to check, but I do see that apt is updating from Trusty repositories, which is where it's finding realpath (as per the output of `apt-cache policy realpath`). Since I'm on Bionic, I'm not sure where or when Trusty repos were added to my machine. I think if it's tested on a new machine, it will become clear if it might be something AP is adding, or if my problems are being caused from some other software (perhaps ROS).

Also -- even if it is my fault to have Trusty repositories confusing the setup script --  I was wondering if we might be able to prevent other people from seeing this bug with this check anyway?

```bash
function is_installed() {
     dpkg --verify "$1" 2>/dev/null
}

# Check if we need to manually install realpath
# Skip if coreutils present
RP=$(apt-cache search -n '^realpath$')
if [ -n "$RP" ] && ! is_installed "coreutils"; then
    BASE_PKGS+=" realpath"
fi
```

It would seem better to check if realpath is needed this way rather than to simply see if it's available! But let me know. And thank you! 😊

nabeelsherazi, 2021-01-09:
Just saw your follow up comment -- it's also definitely possible it's a holdover from 16.04. And if you're not seeing it on a fresh machine, then definitely on my end, I apologize!

khancyr, 2021-01-09:
No need for more code, if you get mixed ubuntu repo version then you got already a big issue that we shouldn't try to address.

closed 2021-01-09 by khancyr