ArduPilot/ardupilot #16310: Copter: AutoTune's position hold can lead to suboptimal tune

rmackay9, 2021-01-13:
Copter's [AutoTune](https://ardupilot.org/copter/docs/autotune.html) feature attempts to weakly hold position if invoked from Loiter mode.  After a discussion with @lthall though it is clear that this is less likely to produce a good tune compared with invoking AutoTune from AltHold (or some other mode) so that it does not attempt position hold.

The reason is that the position hold feature may incorrectly align the copter's yaw so that it is twitching into the wind (or at least not perpendicular to the wind).  This happens because AutoTune was (at some point) enhanced to twitch twice (or more?) in the same direction.  When this happens the vehicle can build up significant horizontal speed in the direction of the twitch and move more than 5m from its start position.  When this happens the vehicle's yaw is changed but the change is based on the assumption that it is the wind pushing the vehicle to the edge of the circle, not the twitch itself.

[image]

robustini, 2021-01-13:
In fact it is always better to run AutoTune from Stabilize or AltHold.

MallikarjunSE, 2021-01-14:
Althold gave me good results all the time.