ArduPilot/ardupilot #16322: Quadplane/Tailsitter: motors stopped for one second during QRTL (bug,

robustini, 2021-01-14:
After an in-depth analysis on some logs detected in the real flight, I also searched for correspondence on the SITL and Realflight, and I can confirm that 'it's a big problem on QRTL with the tailistter VTOL, especially on non-vectored ones.
The problem is evident in the first image that I'm attaching, we note that the throttle remains at minimum (which means motors off) for one second, immediately after the tailsitter exits the condition of angle reached, that is when it virtually switches from plane to copter.
This obviously does not happen for example if i switch from Cruise to Q_Loiter (second image), where, therefore, the transition is completed there is no need to automatically place the tailsitter on the home and then execute the land.
This is pretty boring, as on some tailsitters it makes it impossible to perform a QRTL, so any failsafees may fail as well.
The tailsitter advances in a sort of harrier, no longer able to raise the nose and if the pilot does not intervene manually something unpleasant could happen.
So I ask @IamPete1 to look for where the bug is.

robustini, 2021-01-14:
The problem is even more evident by analyzing the pwm sent to the ESCs, always comparing QRTL with QLoiter.

[image]

[image]

IamPete1, 2021-01-14:
There is no bug, its simply that descending needs low throttle and at low throttle we have no control. Its unrelated to the transition, simply that after the transition it asks for a fast decent.

For a copter frame we know how much throttle we need because roll, pitch and yaw are actuated directly by throttle. So we can boost up the throttle to regain control power. Because tailsitter vectoring/control surfaces are not directly a factor of throttle we cannot judge when/how much to boost the throttle (unless your wingtra and you can characterise your airframe). 

The short term fix is to increase SPIN_MIN such that there is always control power.

We could also use the the disk-theory gain scaling methodology to impose a minimum airspeed over the control surfaces, for hover this would be equivalent to a higher min throttle, but it would allow lower throttles with some airspeed. 

I have tested using pitch and yaw output saturation to trigger increasing throttle but this does not work. The vehicle will tend to descend OK if its level and gain speed, then a small error causes some control surface movement, the vehicle now essentially trying to fly backwards and rapidly diverges (typically in pitch). The boost kicks in and saves it, but only after a large error, and the cycle repeats. 

The trick to doing faster descents is just to keep some forward speed, ideally some wind will do this for you, or we could attempt some sort of spiral decent, like loiter to alt but with a much smaller radius.

Open to better ideas for triggering boost throttle, my testing of using saturated outputs is not promising, using angle error will result in instant rapid takeoffs on arming for TVBS's.

There may be a better solution, but I cant think of it at the moment.

robustini, 2021-01-15:
Thanks @IamPete1 for reply!
Well, then we have identified the problem, and this is a good starting point.
In my opinion the solutions could be many, we need to understand how to implement them.
If in QLoiter all transitions are perfect it means in my opinion that in QRTL something is not done correctly, or rather it is done correctly for a quadplane where the horizon reference is always the same, but not for a tailsitter.
The problem is that in the event of a failsafe at this moment I have to assume that my tailsitter would go to the ground destroying itself, without me being able to do anything.
The only thing I could do is via telemetry to ask for a flight mode change in QLoiter followed by a QLand.
So this is not a marginal problem, but a big problem.
Whether the code is capable of automatically not keeping the drone within the flight envelope in case of failsafe is not irrelevant, but dangerous.
What I mean is that my concern is that I may be in this condition, and I have no way out at the moment.
For example, I cannot run a lua script in the event of a failsafe, where I could make the drone perform a sequence of things to make it land intact, this too would be a very interesting feature.
Something we have to do or something unpleasant might happen to someone.

robustini, 2021-01-15:
@IamPete1 howeverh I found this: the problem is also related to the value of Q_ANGLE_MAX.
In the previous tests I had set it to 3000, which is a reasonable value for this tailsitter.
Now I have tried to set it to only 1000 and on the simulator obviously the ramp changes because the angle to be reached is higher.
In this context the QRTL transitions at least on the simulator perform them really much better, so I also trust in reality.
But obviously I can't leave this angle at only 10°, and that lack of throttle shouldn't be there anyway.
So we could simply solve it by having the possibility, with a parameter, to change the desired pitch angle during the plane-> copter transitions, but which then does not affect the Q_ANGLE_MAX, so only related to this condition.

[image]

robustini, 2021-01-21:
Here the log where the problem is evident.
I change the Q_ANGLE_MAX from 3000 to 500 during the flight, then activate QRTL and the transition happens without problem.

[2021-01-19 11-39-29.zip](https://github.com/ArduPilot/ardupilot/files/5847622/2021-01-19.11-39-29.zip)

pierreletullier, 2021-01-22:
Experience from my blue TVBS
Transition to Q_loiter perform always very well. That is very positive and encouraging.
I did not test QRTL but transition VTOL within a mission: sometime transition is OK but sometime not. The difference seems to be related to requested attitude. Sometime, once the transition angle is reached, the wing yaw rapidly at the same time it pitches up giving unstable attitude.
One the transition is done, auto VTOL flight mode is not very stable (in case of wind) because there is a competition between yaw, pitch and position holding. But I believe this is more a problem of airframe limitation. 
Mission should be adapted: not to enter VTOL transition too fast, first waypoint altitude after transition to be set at the altitude got after transition...
Maximum down velocity set to 100 cm/s, this is the max my wind handle in auto mode. 
Q_m_spim min is very very important.

robustini, 2021-01-22:
Thanks @pierreletullier for your report.
I have also ascertained these things, I have done several tests with the spin min and the tuning on the acceleration, no good results.
Some solutions are possible, but we have to get your hands on the code to implement them.

robustini, 2021-01-23:
> Mission should be adapted: not to enter VTOL transition too fast, first waypoint altitude after transition to be set at the altitude got after transition...

The problem is not carrying out a mission but in case of QRTL, so for example if something happens that triggers the failsafe.
As I had already written above, if a problem occurs on the tx or the rx the crash is almost guaranteed in this condition, which is why the problem is not marginal.
Of course, we can not activate the fialsafe but it is illogical in a safety context.

pierreletullier, 2021-01-24:
yes, I will try Q_rtl and post the log. Parameters are already uploaded I just wait for rain to stop.

robustini, 2021-01-25:
@pierreletullier today we did about ten QRTL by varying the SPIN_MIN, I got to 0.5 (out of scale) considering that the thrust hover is 0.41.
Out of 10 transitions he made one good, all the others failed.

IamPete1, 2021-01-25:
Just changing `SPIN_MIN` will also mess up your throttle expo and PIDs, you will have to change the expo and re-tune if you change it significantly.

robustini, 2021-01-25:
> 
> 
> Just changing `SPIN_MIN` will also mess up your throttle expo and PIDs, you will have to change the expo and re-tune if you change it significantly.

@peterbarker you're right, I'm sorry, I had obviously changed the expo too.

jmachuca77, 2021-01-26:
Maybe I'm over simplifying this but why not delay the descent for a configurable amount of time so that the descent is not requested right at the transition?
We have also noted on the tbvs that the QRTL transitions are quite unstable and the plane basically stalls and then saves itself in quad mode

robustini, 2021-01-26:
> Maybe I'm over simplifying this but why not delay the descent for a configurable amount of time so that the descent is not requested right at the transition?
> We have also noted on the tbvs that the QRTL transitions are quite unstable and the plane basically stalls and then saves itself in quad mode

It would be a clever thing that I had already proposed, but Ardupilot thinks with another steps during that phase.
I had already proposed the sequence during QRTL: 

1. when it comes within range of the RTL it switches as controller in Q_Loiter and remains there for x seconds (configurable) 
2. after this time it moves to the home maintaining that height, without going down 
3. reached the home begins to descend

It would be the end of all these problems.
I was told that it is not a viable path, if I could get my hands on the code at that level I would have already changed it in this way, because the current logic is fine for a multicopter but not for a tailsitter.
The criticality arises precisely because immediately after the transition the tailsitter immediately repositions itself in X (pitch) going down by Z (altitude), and leaving the transition still tilted in pitch it loses its attitude.

pierreletullier, 2021-01-26:
https://www.dropbox.com/s/h93yriy0hq3oxy2/00000269.BIN?dl=0
This is a log of 4 RTL in a row with q_rtl_mode=1. They were successful from my point of view.
Q_wp_speed_dn=100, Q_rtl_alt=20, RTL_radius=15, rally point altitude=15, transition angle=65°, max angle =70°
They were about 5m/s of wind, all RTL were made face to the wind and sometime the wing was not able to go faster than the wind. Altitude gain during transition VTOL is 10 to 15m, that's a lot.  Something bad is the plane heads first to rally point altitude so after transition it is 10 to 15 m above. 
Something I learnt with this tests is to better tune Q_a_rate_rll parameters. This axis is not dampened by air drag and react much faster than other 2 axis which gives a bad roll and unstable dance.  So medium P and I gain and a maximum D gain is better.

jmachuca77, 2021-01-26:
"All of this obviously has no criticality on vectored tailsitters, or in any case it is irrelevant." This is not true Marco, we have a vectored tail sitter and still see this problem on the QRTL transition. It stalls and then catches it self. its really ugly.

robustini, 2021-01-27:
@pierreletullier thanks for the log, I analyze it and do my evaluations. 

@jmachuca77 this is very strange, vectored shouldn't give you this problem if the tuning is good.
The vectored has an edge on angle recovery, provided of course you have thrust necessary to do so.

robustini, 2021-01-28:
> This is a log of 4 RTL in a row with q_rtl_mode=1. They were successful from my point of view.

From what I see in the log you are using much smaller propellers than I am using.
Using 14x8 props like me you will see that many things will change.

pierreletullier, 2021-01-28:
I use 10*5 propeller, I do not use hover learn feature. I adjust Q_m_thst_hover to get the minimum thrust to complete forward transition.

robustini, 2021-01-30:
> 
> 
> I use 10*5 propeller, I do not use hover learn feature. I adjust Q_m_thst_hover to get the minimum thrust to complete forward transition.

Ok, this is a known trick but it's not the solution anyway, as switching to Q_Stabilize your tailsitter in mid throttle will definitely go up, not good.
The transition should also happen with hover learn enabled.

rekomerio, 2021-02-01:
I have done some changes to the code and tested them to fight against this issue. If you'd be interested I could send merge request. https://github.com/rekomerio/ardupilot/blob/dev/reko/ArduPlane/quadplane.cpp#L2311

While I developed this, I also noticed that the plane tries to fly towards next mission point if in mission, while doing the transition and I also fixed that. Let me know if you want me to send merge request

robustini, 2021-02-01:
> I have done some changes to the code and tested them to fight against this issue. If you'd be interested I could send merge request. [...]

Very interesting, thanks!
This is also very usefu, i mean VELZ_MAX_UP and DOWN: https://github.com/rekomerio/ardupilot/commit/f586f210b37afbd2a2f7e5ebbd7ac67d1930a2bf

https://github.com/rekomerio/ardupilot/commit/f586f210b37afbd2a2f7e5ebbd7ac67d1930a2bf

robustini, 2021-02-01:
@rekomerio I tested your fixes on the sim and they are absolutely fine, really thanks!
The operating principle, even from what I see in the code, is in line with what I had proposed as a solution, and in fact it works.
I notice a moment after which completing the transition takes a greater angle to start the repositioning, it does it all of a sudden and it is not nice to see but at least so the transition does it, great!

robustini, 2021-02-02:
Please @rekomerio if you want should do two separate pull requests to the master, one for VELZ_MAX_UP / DOWN and the other one about the fix for transitions.
For both ask for the revision of @peterbarker and @tridge, thanks!