ArduPilot/ardupilot #16326: KakuteF7Mini master copter failing to initialize

tenten8401, 2021-01-15:
When powering on a freshly flashed KakuteF7Mini flight controller (1/14/20 nightly) I'm greeted with either:
> INS: Unable to initialize driver
> Config error: Fix problem and reboot

or

[image]
and it disconnects, always directly after baro calibration.

I don't think it's a flash space issue as I tried building fresh master (today 1/14/20) with a heap of features disabled (autotune, logging, optical flow) and it does the same thing.
[APM_Config.h](https://gist.github.com/tenten8401/b948cc265428f71a34205761c7431aa9)

There is nothing in logging dataflash and the initial flash was done with "full chip erase" in iNav configurator (yeah I know, it's not qgc or mp) via DFU. Custom build flash was done via `./waf copter --upload` and same thing.

I believe the change happened around 1/4/20 as I was helping test a new build and it was doing the same thing, while builds from a few days prior were working fine. Doesn't appear to be related to 6ed4eca though.. (removed added lines and re-built to test, not sure if there's anything else I should try). Builds are only 809932 bytes out of 950272.

andyp1per, 2021-01-15:
I have three of these boards and as far as I know they are all fine. I'll double check tonite, but its worth seeing whether any of my changes are responsible

andyp1per, 2021-01-15:
Maybe try gitblame on the hwdef to see if there is any correllation

tenten8401, 2021-01-15:
I'll see what I can find.. I already looked thru github commit log for the hwdef folder but I'm not finding anything that sticks out.

andyp1per, 2021-01-16:
This works fine for me. I suspect hardware failure. As a double check can you post your parameters for me to load as well?

lavachemist, 2021-02-13:
The newest version (v3) of the Kakute F7 Mini has an F722. Only the previous versions are supported, because they had the F745. Maybe that is your issue?

tenten8401, 2021-02-13:
@lavachemist This was fixed, turned out to be a DMA issue. All good now :)

closed 2021-02-13 by tenten8401