ArduPilot/ardupilot #16343: uavcan library issue

liang-tang, 2021-01-17:
When I was testing the rsc airspeed meter recently, I found that the temperature data is not continuous, usually at an interval of 2 degrees. So I did a series of tests and found that if the value sent is larger, the data interval will be larger, such as 0-20, the minimum interval is 0.1, 20-50, the interval is 0.2, 50-100, the interval is 0.5, 250 or more interval 2(4, 6, 8.....)

tested on commit 26a3a02c818352462d2516abc9f7a54e75bcb84e
tested via below:
[image]

Then I checked the version of uavcan in Ardupilot and found that they are all stuck in 2019. Is it necessary to update the version of the uavcan library?

some data here
[image]
[image]
[image]
[image]

Sypaq-MadMan, 2021-03-24:
Just commenting so I get updates regarding uavcan library.

Sypaq-MadMan, 2021-03-24:
Just read this, I think uavcan v0 is here to stay for a bit. 
https://forum.uavcan.org/t/uavcan-v1-0-and-ardupilot/671/27

magicrub, 2021-03-27:
if uavcan v0 was compatible to what all other hardware on the planet was using, it would have been an easy transition.

IamPete1, 2021-08-16:
It sounds like this is a temperature issue limitation of the protocol, not something that can be fixed in isolation, I'm going to close this.

closed 2021-08-16 by IamPete1