ArduPilot/ardupilot #16367: sim_vehicle.py

FreeCaptain, 2021-01-18:
Hello 
I used SITL:

./waf configure --board sitl   

From any commands with sim_vehicle.py I get:

SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at SITL location
Traceback (most recent call last):
  File "/home/user/ardupilot/Tools/autotest/sim_vehicle.py", line 1277, in <module>
    do_build(vehicle_dir, cmd_opts, frame_infos)
TypeError: do_build() takes 2 positional arguments but 3 were given
SIM_VEHICLE: Killing tasks

**Version**

last version

**Platform**
[ *] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

khancyr, 2021-01-18:
Hello,

Thanks, we are aware of that and it should be fixed soon

peterbarker, 2021-01-18:
This was fixed.  Apologies for the breakage.

closed 2021-01-18 by peterbarker