ArduPilot/ardupilot #16373: start SITL simulator with heli or plane model, cannot takeoff the drone

addriumruss, 2021-01-19:
Today I compile the latest code of ardupilot within cygwin, and start a simulate drone with cmd "usr/bin/arducopter-heli.exe --home 32.1291816,118.8849733,100,0 --model heli --base-port 5760", then connect to it usingMissionPlanner with tcp port. Finally, I want to take off the drone by the "fly here " button in the right click memu, it failed always. 
Is it a bug , or there is anything not right in the way? Anyone can explain this result? Thanks so much~

khancyr, 2021-01-19:
Hello,

What IHM are you using ? Mission planner ? Mavproxy ?

In both case, I pretty sure that you need to set the right mode and trigger a takeoff before being able to use the fly here button.

Please read back the documentation for each vehicle on how to takeoff.

closed 2021-01-19 by khancyr

addriumruss, 2021-01-20:
@khancyr Thanks very much for response~. 
The IHM is Mission planner in windows 10, and it's a SITL build in cygwin.  Indeed I clicked the "Take Off" button in the right click menu at first, then the button "Fly here" was clicked. But the problem is the drone didn't take off actually, and no errors displayed, I really cann't figure out why. With the same ways, the quad drone can be taken off very easy. Is there any more switches with the heli or plane in sitl?