ArduPilot/ardupilot #16403: Automatic landing gear retraction dont work with lidar connected

AckroydPP, 2021-01-21:
## Bug report

**Issue details**

LGR retract alt = 40m, deploy 35m
maximum lidar height is set to 35m 
when lidar is connected and providing data to cube, the LGR never retracts as it should. (it only retracts if i do set retract alt below Lidar maximum height. But again above this height, it deploys again, when it should be retracted) (my conclusions are confirmed with RC10_out logs where it is  connected)
The moment i manually disconnect i2c connector from lidar (no change in parameters) the LGR retracts and deploys as it should perfectly.
ALT_source is set to baro, the lidar is used for IR-lock only in my case.

**Version**
AC 4.0.6-RC1, AC 4.0.5

**Platform**
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadrotor

**Hardware type**
Cube orange, Lidar lite V3

**Logs**
https://drive.google.com/file/d/1Rzi6VIqKeA4HcXWOo5Nl-IsVXJ4mbO8g/view?usp=sharing

AckroydPP, 2021-01-25:
I disovered one thing i dont think it is on purpose and it may relate to this problem.
My settings:
EK2_enable=1
EK2 alt source =0(baro)
lidar is connected
EK2_RNG_USE_HGT tried to set -1/0/1
With all these settings, it for some reason uses lidar as a height source all the time in lower altitudes (testing with large plate moving under the drone so the drone reacts to it and changes altitude accordingly)
I thougt i can have lidar connected just for irlock purposes, but it always use it as height source when it is connected.
Dont now if its related, but you will know for sure :-)

khancyr, 2021-01-25:
Hello,

As per discussion in https://github.com/ArduPilot/ardupilot/pull/9735 that is the intend behaviour. Maybe, we could just update the parameter documentation and maybe in case of case RangeFinder::Status::OutOfRangeHigh check if the get_alt_above_ground_cm > rangefinder_state.alt_cm_filt.get() to use it

AckroydPP, 2021-01-27:
Hello.
thank you for your answer. So if i understand correctly.

1. There is no chance for me (unless i personally change the code) to use lidar for IR-lock ONLY. (alt_source and lgr deploynment based only on baro)

2. If i have a lidar connected, automatic landing gear retraction/deploynment wont work above maximum lidar height.

3. One more question, since it looks like you know the code well. If automatic landing gear retraction is set to be based on height, will landing gear automatically deploy when parachute is deployed automatically? ( You know, to save the camera)

I know i have a lot of questions, but these things are not clear from parameter documentation.
Thank you

khancyr, 2021-01-27:
Hello,

1. No we cannot select which subystem should use rangefinder or not.
2. that is correct
3. in case of parachute, the landing gear will be deployed that doesn't depend on height

AckroydPP, 2021-01-27:
Hello. 
Thank you for your answers. 
I understand it now.
Maybe one more question.
1. Since in my case lidar is the problem, am i able to successfully use PLND (with ir-lock) without lidar? I guess propably not.