ArduPilot/ardupilot #16450: EKF2 instances have different origin height after GPS is aquired when launching without GPS

jfint, 2021-01-27:
## Bug report

**Issue details**

When launching a plane without GPS and allowing it to acquire GPS in flight the Origin Heights of the 2 instances of EKF2 show a different value.

I thought that I was onto something with this check-in but failed to produce the results needed, possibly because this is still a multi-threaded race condition.

https://github.com/jfint/ardupilot/commit/8051636425b8225721bd89dff1548dd37a429bc0

**Version**
Arduplane 4.0.5 and Arduplane 4.0.7

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Flying Wing

**Hardware type**
CUAVv5Nano Clone

**Logs**
_Please provide a link to any relevant logs that show the issue_
[image]

[Flt272.zip](https://github.com/ArduPilot/ardupilot/files/5877026/Flt272.zip)

vierfuffzig, 2021-01-29:
https://github.com/ArduPilot/ardupilot/issues/10812

hendjoshsr71, 2021-06-18:
@jfint you may want to see if the changes here #17786 solves this issue for you.

jfint, 2021-06-18:
@hendjoshsr71 Interesting, we have actually reverted back to the DCM filter, we found about a 20% probability that the EKF filter never converged in GPS denied launches and unfortunately had to look to the older solution as a possible band-aid.  Haven't worked on that in quire a while, will need to revisit soon.

hendjoshsr71, 2021-07-01:
@jfint The PR that fixes this issue got merged so I'm closing this one. 

Sorry to hear about your issues with EKF filter in GPS denied launches. There has been a lot of work in starting with visual odometry and gaining GPS later on. 

You may want to try the soon to be released 4.1. If you do have problems, please make a new issue with logs about those so it can be looked into.

closed 2021-07-01 by hendjoshsr71