ArduPilot/ardupilot #16451: Analog RSSI scaling on Holybro Durandal. Less than 45% of full analog input range is used.

rotorman, 2021-01-27:
## Bug report
Analog RSSI input scaling on [Holybro Durandal](http://www.holybro.com/product/durandalbeta/) needs improvement, as presently only only less than half of the measurement range is used.

**Issue details**
Presently when RSSI_PIN_HIGH is set to 3.3V, all values above approx. 1.36V are handled as 0xFF (255).
When RSSI_PIN_HIGH is set to 5.0V, the threshold above which output is 0xFF is approx. 2.15V.

Expected behavior: 0 to 0xFF should be mapped to full scale from 0 to 3.3V (RSSI_PIN_HIGH set to 3.3V) or 0 to 5.0V (when RSSI_PIN_HIGH is set to 5V).

Following other settings were used for test:
RSSI_TYPE is 1 (=Analog Pin)
RSSI_ANA_PIN is 103 (=Pixhawk SBUS)
RSSI_PIN_LOW is 0

**Version**
ArduPlane V4.0.8 (and also the same with ArduPlane V4.0.7)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
FLYING_WING

**Hardware type**
Holybro Durandal (purchased directly from Holybro)

rotorman, 2021-01-27:
Please also check the handling of 0xFF (=255), as [MAVLink #65](https://mavlink.io/en/messages/common.html#RC_CHANNELS), [MAVLink #35](https://mavlink.io/en/messages/common.html#RC_CHANNELS_RAW) and even [MAVLink #109](I) valid range is 0 to 254 and 0xFF indicator is used for invalid.
This means, the value should in normal case not exceed 254 (0xFE).

rotorman, 2021-02-05:
I got access to the Holybro Durandal original schematics thanks to Holybro support and can confirm that the scaling issue must stem from I/O co-processor (STM32F100) firmware, as the analog input is fed only via 10K resistor to STM32F100 I/O co-processor pin PA5 (physical chip pin 15) and the supply is 3.3V.
According to Holybro support the I/O co-processor can be updated via main FMU H743 processor, thus it should be possible to fix this via ArduPilot fw.

rotorman, 2021-02-11:
I found the culprit to be in the firmware used for the I/O co-processor. Whopping 3 bugs in a row, and opened an issue at PX4:
https://github.com/PX4/PX4-Autopilot/issues/16829

rotorman, 2021-02-11:
Nevertheless, the bug of outputting 0xFF by ArduPilot for full signal is wrong, as according to MAVLink specs 0xFF is reserved for invalid data.
ArduPilot should limit the output of a valid signal to 0xFE=254.

rotorman, 2021-02-15:
PR #16607 should fix this now.

IamPete1, 2021-08-16:
I think the correct fix for the scaling issue is to change the hwdef here:
https://github.com/ArduPilot/ardupilot/blob/8bd04c17c85e8ac2e794771608cde91e7c5660dd/libraries/AP_HAL_ChibiOS/hwdef/Durandal/hwdef.dat#L201

rotorman, 2024-06-16:
Fixed by #18754. Closing.

closed 2024-06-16 by rotorman