ArduPilot/ardupilot #16461: Poor Autotune Performance

lthall, 2021-01-29:
I have been noticing that Autotune has felt lest reliable for a while now. I have been noticing that the twitch tests have been inconsistent and unusually close together. I have looked into this problem and found that this is being caused by the relaxed level test added here:
https://github.com/ArduPilot/ardupilot/commit/a9ec4c342294fc2aaca05488c892a7232866f8e4

From the beginning there has been pressure to speed up Autotune and get autotune to run on badly tuned aircraft. The intention is to help users get to a basic tune without the need to do a manual tune. Unfortunately this has resulted in autotune producing bad results on aircraft that would otherwise have tuned up well.

I would propose that we decrease the maximum angle error to 2.5 degrees and maximum rate error of 5 degrees per second. We could increase these limits up to a factor of 2 after 2 seconds to help poorly tuned aircraft get a starting tune. However doing autotune when an aircraft is not able to maintain less than 5 degrees of level or 10 degrees per second in rate is not likely to achieve good results (or dangerous results) and the pilot should be directed to do a manual tune.

In the event we do not twitch for 5 seconds we may be able to put a message up saying "Unable to level aircraft, manual tuning necessary" or something like that.

Along the lines of messages it may also be worth putting up an error message in the event that the sticks are lot centred for more than 5 seconds. "Autotune waiting for sticks to be centred" or something similar.

peterbarker, 2021-01-29:
On Fri, 29 Jan 2021, Leonard Hall wrote:

> I would propose that we decrease the maximum angle error to 2.5 degrees and maximum rate error of 5 degrees per second. We could increase these limits up
> to a factor of 2 after 2 seconds to help [...]

Would there be utility in having these parameterised?

> Along the lines of messages it may also be worth putting up an error message in the event that the sticks are lot centred for more than 5 seconds.
> "Autotune waiting for sticks to be centred" or something similar.

We already do that?

https://github.com/ardupilot/ardupilot/blob/master/libraries/AC_AutoTune/AC_AutoTune.cpp#L250

lthall, 2021-01-30:
> Would there be utility in having these parameterised?

I don't think so. I don't think I could tell people how to set them.

> We already do that?

Yeh, I was thinking an error message if it stays in that mode but that is just a thought. More significantly would be if the aircraft can't level and looks like it is just sitting there. Maybe we need to abort Autotune and tell the user to go tune their aircraft better.

The messaging ideas were secondary thoughts.

closed 2021-05-04 by lthall