ArduPilot/ardupilot #16469: UAVCAN based GPS is not detected

wucke13, 2021-01-31:
## Bug report

**Issue details**

The GPS turns on and eventually finds a fix (indicated by green LED switching on for a few ms every at roughly 1 Hz). However, it seems like the communication to the GPS is not working at all: Even with a fix, my position as indicated by APMplanner is lat/long = 0,0. No GPS fix is indicated by Ardupilot, and of course 0 satellites are in view. Interesting enough, MissionPlanner does not show any Node in Setup/Optional Hardware/UAVCAN.

Maybe interesting: The blue LED on the GPS flashes with 0.5 Hz, 50% duty cycle.

I did the setup of the GPS according to the docs. This is are the relevant snippets from my `.param` file:

```
CAN_D1_PROTOCOL,1
CAN_D1_UC_ESC_BM,0
CAN_D1_UC_NODE,10
CAN_D1_UC_SRV_BM,0
CAN_D1_UC_SRV_RT,50
CAN_D2_PROTOCOL,0
CAN_P1_BITRATE,1000000
CAN_P1_DRIVER,1
CAN_P2_BITRATE,1000000
CAN_P2_DRIVER,1
CAN_SLCAN_CPORT,0
CAN_SLCAN_SERNUM,-1
CAN_SLCAN_TIMOUT,0
COMPASS_TYPEMASK,0
GPS_TYPE,9
GPS_TYPE2,9
```

**Version**
ArduPlane V4.0.8 (ac9a4b9a) 
ChibiOS: d4fce84e 
MatekH743 003B002A 3139510A 34383432 

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Plane, 2 Motor, 2 Aileron, 2 Flap, Rudder, Elevator

**Hardware type**
Matek H743 Wing
M8Q-CAN

rmackay9, 2021-01-31:
This is really a support issue so it would be best to raise it in the discuss forums first: https://discuss.ardupilot.org/c/arduplane/plane-4-0/153.  I know it can sometimes be hard to get an answer in the forums but it is likely that this is a configuration issue either in the GPS or the AP parameters.

Sorry to be strict but if we're not then the developer to-do list turns into another support forum.

closed 2021-01-31 by rmackay9

wucke13, 2021-02-01:
We will see. I would be happy to find out that I did wrong. If all parameters where set according to the docs, and it's still not working, what would be the indication that it may be a bug?

[Discussion](https://discuss.ardupilot.org/t/uavcan-based-gps-is-not-detected/66995)

wucke13, 2021-02-01:
The issue is solved. The problem was nasty: The CAN-L and CAN-H on the cable subpplied by Matek with the M8Q-CAN where flipped on the FC side. Therefore no successful CAN communication. @rmackay9 Thank you for the gentle strictness - it turned out to be correct.