ArduPilot/ardupilot #16474: Terrain not working on Pixhawk 4 FC

Hasi123, 2021-02-01:
## Bug report

All terrain following features on the Pixhawk 4 don´t work. Terrain data is loaded from the GCS and the copter displays the current height AGL correctly through telemetry and logs, but it won´t use this data in terrain following modes (auto, rtl). Instead it uses above takeoff heights.

**Version**
tested on 4.0.5 and 4.0.6

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadcopter

**Hardware type**
Pixhawk 4

**Logs**
Forum thread and log can be found here: https://discuss.ardupilot.org/t/terrain-following-not-working-on-pixhawk-4/66789

Yatin-Khurana, 2021-02-02:
Have you enabled obstacle avoidance ? (OA_TYPE)

rmackay9, 2021-02-02:
Yes, @Yatin-Khurana is correct that the issue is likely this known issue with object avoidance and terrain following: https://github.com/ArduPilot/ardupilot/issues/16479

closed 2021-02-02 by rmackay9