ArduPilot/ardupilot #16476: MRO Control Zero won't build on Master

ryanjAA, 2021-02-02:
I can build other boards without issue (Cube, etc) on todays and (back several commits) Master and but not the MRO Control Zero F7.

On macOS 10.15.4

```
ryandev@ryandevs-MacBook-Pro ardupilot % ./waf configure --board mRoControlZeroF7                                           
Setting top to                           : /Users/ryandev/src/ardu_master_feb1/ardupilot 
Setting out to                           : /Users/ryandev/src/ardu_master_feb1/ardupilot/build 
Autoconfiguration                        : enabled 
Setting board to                         : mRoControlZeroF7 
Using toolchain                          : arm-none-eabi 
Checking for 'g++' (C++ compiler)        : /usr/local/bin/arm-none-eabi-g++ 
Checking for 'gcc' (C compiler)          : /usr/local/bin/arm-none-eabi-gcc 
Checking for c flags '-MMD'              : yes 
Checking for cxx flags '-MMD'            : yes 
CXX Compiler                             : g++ 9.3.1 
Checking for program 'make'              : /usr/bin/make 
Checking for program 'arm-none-eabi-objcopy' : /usr/local/bin/arm-none-eabi-objcopy 
Adding environment OPTIMIZE -O2
Setup for MCU STM32F777xx
Writing hwdef setup in /Users/ryandev/src/ardu_master_feb1/ardupilot/build/mRoControlZeroF7/hwdef.h
MCU Flags: cortex-m7 ['-mcpu=cortex-m7', '-mfpu=fpv5-d16', '-mfloat-abi=hard']
Writing DMA map
Generating ldscript.ld
Default parameters path from hwdef: /Users/ryandev/src/ardu_master_feb1/ardupilot/libraries/AP_HAL_ChibiOS/hwdef/mRoControlZeroF7/defaults.parm
Checking for env.py
env set OPTIMIZE=-O2
env set WITH_FATFS=1
env set PROCESS_STACK=0x1C00
env set MAIN_STACK=0x600
env set IOMCU_FW=0
env set PERIPH_FW=0
env set HAL_NUM_CAN_IFACES=1
env set BOARD_FLASH_SIZE=2048
env set CPU_FLAGS=['-mcpu=cortex-m7', '-mfpu=fpv5-d16', '-mfloat-abi=hard', '-DARM_MATH_CM7', '-u_printf_float']
env set CORTEX=cortex-m7
env set APJ_BOARD_ID=141
env set APJ_BOARD_TYPE=STM32F777xx
env set USBID=0x1209/0x5740
env set FLASH_RESERVE_START_KB=96
env set FLASH_TOTAL=1998848
env set DEFAULT_PARAMETERS=/Users/ryandev/src/ardu_master_feb1/ardupilot/libraries/AP_HAL_ChibiOS/hwdef/mRoControlZeroF7/defaults.parm
env set CHIBIOS_BUILD_FLAGS=USE_FATFS=yes CHIBIOS_STARTUP_MK=os/common/startup/ARMCMx/compilers/GCC/mk/startup_stm32f7xx.mk CHIBIOS_PLATFORM_MK=os/hal/ports/STM32/STM32F7xx/platform.mk MCU=cortex-m7 ENV_UDEFS=-DCHPRINTF_USE_FLOAT=1
Traceback (most recent call last):
  File "/Users/ryandev/src/ardu_master_feb1/ardupilot/modules/waf/waflib/Scripting.py", line 158, in waf_entry_point
    run_commands()
  File "/Users/ryandev/src/ardu_master_feb1/ardupilot/modules/waf/waflib/Scripting.py", line 251, in run_commands
    ctx = run_command(cmd_name)
  File "/Users/ryandev/src/ardu_master_feb1/ardupilot/modules/waf/waflib/Scripting.py", line 235, in run_command
    ctx.execute()
  File "/Users/ryandev/src/ardu_master_feb1/ardupilot/modules/waf/waflib/Configure.py", line 159, in execute
    super(ConfigurationContext, self).execute()
  File "/Users/ryandev/src/ardu_master_feb1/ardupilot/modules/waf/waflib/Context.py", line 204, in execute
    self.recurse([os.path.dirname(g_module.root_path)])
  File "/Users/ryandev/src/ardu_master_feb1/ardupilot/modules/waf/waflib/Context.py", line 286, in recurse
    user_function(self)
  File "/Users/ryandev/src/ardu_master_feb1/ardupilot/wscript", line 307, in configure
    cfg.get_board().configure(cfg)
  File "Tools/ardupilotwaf/boards.py", line 45, in configure
    self.configure_env(cfg, env)
  File "Tools/ardupilotwaf/boards.py", line 568, in configure_env
    cfg.load('chibios')
  File "/Users/ryandev/src/ardu_master_feb1/ardupilot/modules/waf/waflib/Configure.py", line 270, in load
    func(self)
  File "Tools/ardupilotwaf/chibios.py", line 365, in configure
    setup_canmgr_build(cfg)
  File "Tools/ardupilotwaf/chibios.py", line 262, in setup_canmgr_build
    cfg.srcnode.find_dir('modules/uavcan/libuavcan/include').abspath(),
AttributeError: 'NoneType' object has no attribute 'abspath'

```

auturgy, 2021-02-02:
G'day @ryanjAA 
Strange.  The firmware build server isn't failing (https://firmware.ardupilot.org/Plane/latest/mRoControlZeroF7/), and I've just built master locally to check:
```BUILD SUMMARY
Build directory: /home/james/Development/ardupilot/build/mRoControlZeroF7
Target         Text     Data  BSS     Total  
---------------------------------------------
bin/arduplane  1476543  2408  391016  1869967

Build commands will be stored in build/mRoControlZeroF7/compile_commands.json
'plane' finished successfully (2m32.019s)```

I'd look at compiler version - we're not using gcc9 in the dev or build envs.  Env scripts are here: https://github.com/ArduPilot/ardupilot/tree/master/Tools/environment_install

ryanjAA, 2021-02-02:
Thanks. I’m going to close this as it could be built on a Windows computer and is probably setup specific but I’m new to the Ardupilot side so still teething :) 

thanks for looking at it.

ryanjAA, 2021-02-02:
And g’day back to ya 👍 :)

closed 2021-02-02 by ryanjAA

auturgy, 2021-02-02:
For dev issues feel free to jump on discord https://ardupilot.org/discord and ask questions - the team is usually there.

ryanjAA, 2021-02-02:
Awesome - thanks. That will be easier than opening an issue. Much appreciated.

peterbarker, 2021-02-02:
I replicated this by cloning the ardupilot repository but not updating its submodules.

You can fix this with a `git submodules update --recursive  --init`

Which instructions were you following that led you to this failure - we should fix them.

ryanjAA, 2021-02-02:
Success - that fixed it. Thanks a bunch!