ArduPilot/ardupilot #16478: Copter: 4.1.0 issues list

rmackay9, 2021-02-02:
This is a list of known issues especially blockers for the 4.1.0 copter release.

Blockers before start of beta testing:

- [ ] S-curves PR: https://github.com/ArduPilot/ardupilot/pull/15896

Reports requiring investigation:

- [ ] mRo Pixracer Pro gyro & accel health failures + GSF constrain nan internal error (ask Randy for logs)

andyp1per, 2021-02-02:
@rmackay9 from my perspectve its pretty important that https://github.com/ArduPilot/ardupilot/pull/15984 goes in

IamPete1, 2021-02-02:
I think we should also bring scripting up to date with master. Mostly just bindings and one or two helpers.

rmackay9, 2021-02-02:
@IamPete1, 4.1 will branch from master so it will have everything in master.

amilcarlucas, 2021-02-03:
I vote for GPS ordering bugfix and AC_fence support PRs

rmackay9, 2021-02-03:
@amilcarlucas, @andyp1per, I've added two PRs mentioned to a new "before stable release" section.  We can expect the beta period will be up to 3 months (depending upon how quickly we investigate and resolve issues) and we often rebase on master for the first few release candidates so I think there is time to get these into the stable release even if they don't make it into the first beta release.

amilcarlucas, 2021-02-04:
The AC_Fence stuff is https://github.com/ArduPilot/ardupilot/pull/15288

joshanne, 2021-02-05:
@rmackay9 #15288 (AC_Fence for Plane) is in a state ready for people to look at, review, comment, etc.

pompecukor, 2021-02-07:
@rmackay9 @IamPete1 @pkocmoud 
With the help of Peter Hall, we will try to investigate the Pixracer Pro unhealthy gyro issue on 4.1 (3rd item on the above list).
Peter has suggested to see if turning off/on each IMU will help identify the source. I will do that tomorrow and report here.

anbello, 2021-03-20:
I think also #16842 should go in.

amilcarlucas, 2021-03-22:
@rmackay9 the Fence has been merged now. 4.1 will have low altitude fences, and ArduPlane Fences will work in QGC.

JoshWelsh, 2021-04-14:
Hi @rmackay9 I see that the S-Curve "climb-away" has been marked as resolved, but I was curious if there was a discussion somewhere that detailed what the fix/finding was?  Thank you!

rmackay9, 2021-04-14:
Hi @JoshWelsh, the issue was a rally point had been setup with an altitude of 90m so because that point was closer than home the copter started climbing to get to that alt (or maybe 15m above that alt).  I reproduced the behaviour both with 4.0 and 4.1 and it's unchanged.  Climbing to 90+ meters is scary if you're not expecting it but the copter would have been OK and would have gotten to the rally point safely.

peterbarker, 2021-04-15:
On Wed, 14 Apr 2021, Josh wrote:

> Hi @rmackay9 I see that the S-Curve "climb-away" has been marked as resolved, but I was curious if there was a discussion somewhere that detailed what the fix/finding was? Thank you!

User (me) error :-)

JoshWelsh, 2021-04-16:
> Hi @JoshWelsh, the issue was a rally point had been setup with an altitude of 90m so because that point was closer than home the copter started climbing to get to that alt (or maybe 15m above that [...]

Great thanks Randy!  Great to hear it was an easy find! 

> On Wed, 14 Apr 2021, Josh wrote: Hi @rmackay9 I see that the S-Curve "climb-away" has been marked as resolved, but I was curious if there was a discussion somewhere that detailed what the fix/finding was? Thank you!
> User (me) error :-)

You should see the stuff I've done!  Glad you got the copter back.

lthall, 2021-05-10:
#17345 and #17399

Hwurzburg, 2021-05-19:
-[ ] new bootloader fails on trad heli firmware

andyp1per, 2021-06-09:
MOT_PWM_MIN/MAX not being set to 1000/2000 when MOT_PWM_TYPE is set to a DShot type, but SERVO_BLH_* parameters not setup. Looks like a regression from 4.0 is now fixed

andyp1per, 2021-06-09:
Extant Dshot issues:

- [ ] Bi-directional dshot not working on MatekF405 [discussion](https://discuss.ardupilot.org/t/bi-directional-dshot-support/70210/55)
- [ ] DShot doesn't work on KakuteF7 with default settings [discussion](https://discuss.ardupilot.org/t/copter-4-1-0-and-kakute-f7/71875/9)
- [ ] BLHeli passthru not working with Pixhawk 2.4.8 with TMotor 45A [discussion](https://discuss.ardupilot.org/t/t-motor-45a-2-6s-v2-4in1-connected-to-pixhawk-2-4-8/71675/37)

lthall, 2021-06-20:
low LOIT_ACC_MAX leads to incorrect loiter speed (issue) Fix here: https://github.com/ArduPilot/ardupilot/pull/17815
 Guided with high rate position target updates don't work (see issue #17795) Fix here: https://github.com/ArduPilot/ardupilot/pull/17539

andyp1per, 2021-07-01:
RunCam split 4 video start/stop not working [discussion](https://discuss.ardupilot.org/t/control-runcam-split-4-video-recording-from-a-radio-transmitter/62439/126)

rmackay9, 2021-07-01:
@andyp1per, txs for this.   Maybe keep a single list in your comments from 22 days ago (a few comments higher in this issue..)

lthall, 2021-08-02:
#18190 to address the lean angle limit for auto.

rmackay9, 2021-09-04:
@pompecukor @IamPete1 @pkocmoud,

I've marked the "mRo Pixracer Pro gyro & accel health failures" issue as resolved above because in a quick test by Tridge we couldn't reproduce the issue.  This is a very old issue from early on in the beta and I haven't heard anything about it since so I suspect it was a one-off problem with just one user's autopilot or we've resolved the issue as part of other fixes in subsequent beta releases.

Of course I'm very happy to re-open the issue if anyone can reproduce it with a newer beta.

hendjoshsr71, 2022-01-25:
I checked off one that i fixed via, https://github.com/ArduPilot/ardupilot/pull/19263 "AP_Periph HAL battery monitor default for Matek board (discussion, discussion2)"