ArduPilot/ardupilot #16479: Terrain Following not Working with OA_TYPE=1

Yatin-Khurana, 2021-02-02:
## Bug report

Issue : Terrain Following in Auto mode doesn't work if OA_TYPE is set to 1. Although the same setup works in Loiter Mode without any issue.

Description:
Hi, I am trying to achieve Terrain Following on a Hexcopter using a rangefinder(tf03), everything works fine when the oa_type is 0, I tested it on the same mission plan (with terrain altitude type) and the vehicle follows the rangefinder input, but as soon as I switch on oa_type to 1 (haven't tested with 2) the vehicles doesn't use rangefinder(down) for the same mission that worked earlier, instead the vehicle follows the barometer input.
Please Note, that the vehicle- both follows the terrain and use Avoidance in loiter Mode. and The Setup has Two Saperate Rangefinders one with orient 25 and one with 0.

Version
ArduCopter 4.0.4

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

Airframe type
Hexacopter

Hardware type
CubeOrange with Kore Karrier Board

Logs
https://drive.google.com/file/d/1K9EDu0XW4Q8KL5uSnjJjBlntYfpREzo2/view?usp=sharing

Yatin-Khurana, 2021-02-03:
So is anyone working to resolve this currently ?

rmackay9, 2021-03-15:
@Yatin-Khurana, I think I've created a fix for this issue now.  see PR: https://github.com/ArduPilot/ardupilot/pull/16901

If you're able to test it that would be great but no pressure of course.

rmackay9, 2021-03-17:
This should be fixed in master now so closing.

closed 2021-03-17 by rmackay9

Yatin-Khurana, 2021-03-22:
Thanks a lot, I'll try it soon.