ArduPilot/ardupilot #16481: Do change Acceleration

khancyr, 2021-02-02:
## Feature request

**Is your feature request related to a problem? Please describe.**
As  we got a message to change the current speed, we should support a message to change the current acceleration

**Describe the solution you'd like**
An earlier try was made in https://github.com/ArduPilot/ardupilot/pull/9576 but it was hacking the do_change_speed message.
Therefore, this would need a new mavlink message similare to the do_change_speed one's to do the acceleration change.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[ X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

rmackay9, 2021-02-02:
I wonder how useful this change is though really.  How many users really need to actively change how quickly the vehicle accelerates during a mission?  I don't think we should add features just because we can, we really need at least one or more users to describe why they would want this.  Sure, we can dream up an edge case where it is useful but at the same time we need to recognise the ongoing support and dev cost and make sure the feature is useful enough to justify the cost.

Gone4Dirt, 2021-02-03:
We do have a user that is currently struggling with this.  The aggressive deceleration of the vehicle is concerning operators and the general public near the vehicle as it stops at a given location.  Higher accelerations give the perception of being aggressive/twitchy, which a number of operators have mentioned to me that they do not like.  The acceleration can be reduced by reducing WPNAV_ACCEL.  However, doing so messes with the leash calculation and can lead to waypoint acceptance not being very smooth.  So I am in favour of giving users the ability to reduce acceleration without messing with the leash length calculation.
Do I understand correctly, that s-curves will remove the leash calculation?  If so, I don't see the need for a do-change-acceleration command.
If we think there is value in the do_change_accel command, I am happy to implement.  Just don't want to duplicate effort if s-curves fixes this issue.

khancyr, 2021-02-03:
I got some contact that are doing acc based control ... so that may interest them ... but that a pretty small edge case.

rmackay9, 2021-02-03:
@Gone4Dirt, yes, the s-curves removes the leash and I think this will resolve the issue you mention (although I haven't seen it personally).

rmackay9, 2021-02-07:
@Gone4Dirt, I suspect the issue you're seeing is a tuning issue but if you have a log we can confirm in any case.

I think I'll close this until a user specifically asks for it.

closed 2021-02-07 by rmackay9