ArduPilot/ardupilot #16491: Plane: LOITER_TIME cause quadplane to crash.

aabs7, 2021-02-03:
## Bug report

**Issue details**

We are trying to check LOITER for quadplane. The mission we've given is ``` VTOL_TAKEOFF (10 m) , LOITER_TIME(10m, 15 secs) , LAND```. The quadplane takesoff and changes to loiter mode when it wobbles and crashes. 

We've set ```Q_GUIDED = 1```.

**Version**
Arduplane 4.0.1

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad X

**Hardware type**
Pixhawk

**Logs**
[tlog](https://drive.google.com/file/d/17QrI4yHRKgXAzBFGlqWbwrbbol_nkJ2V/view?usp=sharing)

**Other link**
[parameter](https://github.com/aabs7/Quadplane_parameters/blob/master/gauthali.param),
[video](https://drive.google.com/file/d/10wqtVvuokWVUQfW5e3nv6-8lDoj7-sMW/view?usp=drivesdk),

robustini, 2021-02-03:
Please post the dataflash log file, that tlog is unusable.

robustini, 2021-02-03:
Your mission is conceptually wrong. 
Your mission is conceptually wrong, there is no bug in Ardupilot that can cause this crash.
Better read our documentation thoroughly.

https://ardupilot.org/plane/docs/quadplane-auto-mode.html

robustini, 2021-02-03:
Your mission is conceptually wrong, there is no bug in Ardupilot that can cause this crash.
Better read our documentation thoroughly.

https://ardupilot.org/plane/docs/quadplane-auto-mode.html

auturgy, 2021-02-04:
I'm going to close this, as from the information provided it is likely a user setup problem, not a bug or issue.  Please open a thread in our forum (discuss.ardupilot.org) to work through your setup.

closed 2021-02-04 by auturgy

peterbarker, 2021-02-11:
@aabs7 dataflash log?