ArduPilot/ardupilot #16495: Loiter turn radius.

khancyr, 2021-02-03:
## Bug report

**Issue details**

Currently we implement the loiter_turn radius as `uint8_t circle_radius_m = HIGHBYTE(cmd.p1); // circle radius held in high byte of p1` see  https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/mode_auto.cpp#L1234 for example. 
In the mavlink specification we got P3 = loiter radius : https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_LOITER_TURNS

Therefore we should use P3 

**Version**
_What version was the issue encountered with_

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

muramura, 2021-02-03:
@khancyr san, @rmackay9 san.
I have aggregated parameters 1 and 3 of LOITER_TURNS internally into command 1.
If you want to use parameter 3, you need to modify the following process.

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mission/AP_Mission.cpp#L867

khancyr, 2021-02-03:
Thanks @muramura that was what I was missing ! in auto we are using AP_Mission::Mission_Command and not the raw mavlink cmd ... So it make sens. I don't think there is need to extend the Content union to get the heading parameter from the Loiter_turn as we don't have demand for it.

Closing this !

closed 2021-02-03 by khancyr