ArduPilot/ardupilot #16499: Error in AC_WPNav::advance_wp_target_along_track

RoelHelsen, 2021-02-03:
## Bug report
**Issue details**
I think I found a logical error in the AC_WPNav::advance_wp_target_along_track(float dt) function.
`   // get up leash if we are moving up, down leash if we are moving down`
`   float leash_z = track_error.z >= 0 ? _pos_control.get_leash_up_z() : _pos_control.get_leash_down_z();`
Here, to my current understanding, track_error.z is positive when the vehicle is above the track and should move down and the error is negative when the vehicle is below the track.
In the current logic, leash_up is selected when the vehicle is above the track and leash_down is selected when the vehicle is below the track. Using these leash lengths, the velocity limiting performed afterwards still works but not as intended, I think.

I came across this when simulator testing altitude holding with terrain following (using the terrain files) and noticed that the copter would slow down more strongly when the terrain under the copter was rising (and the vehicle is below track) than when the terrain was dropping (and vehicle above track).

**Version**
Copter-4.0.5
still present in master as far as I know

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
all

**Hardware type**
all

**Logs**
[logs.zip](https://github.com/ArduPilot/ardupilot/files/5919410/logs.zip)
[image]
Attached are two log files of a nearly identical mission over a hill. In log 52 the current logic is used and in log 53, the logic is reversed.
`   float leash_z = track_error.z < 0 ? _pos_control.get_leash_up_z() : _pos_control.get_leash_down_z();`

rmackay9, 2021-06-07:
Thanks for the report, this probably was a bug and could lead to the vehicle climbing or descending too slowly.  This code is gone now though in 4.1 so hopefully the issue has been inadvertently solved.  I think we might as well close this because it is unlikely we will release another 4.0 point release.

RoelHelsen, 2021-06-10:
You're welcome. I didn't look at Copter 4.1 yet. Do you have any idea when 4.1 will be released?

rmackay9, 2021-07-30:
@RoelHelsen,

Sorry, I missed your reply somehow.  Copter-4.1.0 is in beta testing now so anyone is more than welcome to give it a try. It is hard to estimate when it will become the stable release but perhaps in another 4 to 6 weeks.

rmackay9, 2021-07-30:
I think I will close this if that's ok because we are very unlikely to go back and do another 4.0 point release (e.g. 4.0.8).  Txs again for the report and investigation.

closed 2021-07-30 by rmackay9

RoelHelsen, 2021-08-01:
@rmackay9 

> @RoelHelsen,
> 
> Sorry, I missed your reply somehow. Copter-4.1.0 is in beta testing now so anyone is more than welcome to give it a try. It is hard to estimate when it will become the stable release but perhaps in another 4 to 6 weeks.

Looking forward to it, I'm curious what improvements we will see in Copter 4.1

rmackay9, 2021-08-02:
@RoelHelsen,

Great.  By the way 4.1.0-beta6 has been released last week so anybody can give it a try when they are ready.  No pressure of course but being the 6th beta release now it is relatively stable and reliable.