ArduPilot/ardupilot #16509: COMPASS_USE=0, COMPASS_USE2=1 results in no compass being used

rmackay9, 2021-02-04:
I've discovered that if COMPASS_USE=0 then no compass data will be fused into EKF3 (and perhaps other AHRSs).
[image]

An example is shown above which I think from a user point of view could be considered reasonable.  So in this situation the 1st compass is set to not-be-used.  2nd compass is set to be used.  EKF3 will not use the compass however.

FYI @bugobliterator @tridge

bugobliterator, 2021-02-04:
Why not make the secondary compass as primary instead. What do you think should be the behaviour in this circumstance. Primary compass in this particular case has been disabled. Remember you are not disabling "Compass 1" here but primary compass.

Previously also I think setting COMPASS_USE 0 led to compass not getting fused.

rmackay9, 2021-02-04:
@bugobliterator,

A user could change the priority but they could also think that it's not necessary.  It's perfectly reasonable for a user to think that if they specify not to use the primary compass that the secondary compass will be used (because they've specified the 2nd compass should be used).
The way it is now, we are ignoring the user's specification that the secondary compass be used.

I think we should add a pre-arm check and ensure in the wiki that this unexpected behaviour is clear.

By the way, what happens in this case?

compass-use1 = 1
compass-use2 = 0
compass-use3 = 1

Will the 3rd compass ever be used?

rmackay9, 2021-03-19:
Small update that I got a user support call based on this issue today.  We need to add a warning or something so this doesn't happen.

bugobliterator, 2021-03-31:
@Hwurzburg its not currently supported. Current process is to swap the order and move compass 1. This PR is going to change that though https://github.com/ArduPilot/ardupilot/pull/17064

IamPete1, 2021-08-16:
similar to https://github.com/ArduPilot/ardupilot/issues/10785

rmackay9, 2021-08-16:
I think we can close this issue now.

closed 2021-08-16 by rmackay9