ArduPilot/ardupilot #16512: Copter: PreArm Battery low voltage failsafe message repeated two times

giacomo892, 2021-02-04:
## Bug report

On Copter, **PreArm: Battery 1 low voltage failsafe** is always sent 2 times to the GCS each time it is broadcasted:

[image]

TLOG:

[2021-02-01_21-29-00.tlog.zip](https://github.com/ArduPilot/ardupilot/files/5924133/2021-02-01_21-29-00.tlog.zip)

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Matek F405

rmackay9, 2021-02-04:
I've noticed this as well.  I thought that I had already added it to the issues list but I don't see it anywhere.

WickedShell, 2021-02-04:
This is an easy fixup, it's due to copter duplicating the battery checks here: https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/AP_Arming.cpp#L141 that should just be removed, and be relying upon the parent class to run the battery checks... (Unless copter want's to override the whole `battery_checks` call explicitly).

sauntheninja2, 2021-02-05:
I want to work on this issue removing that extra check should solve the problem right?

WickedShell, 2021-02-08:
#16530 was merged, closing.

closed 2021-02-08 by WickedShell