ArduPilot/ardupilot #16532: RTL causes the craft to ascend and then immediately descend to RTL alt

skorokithakis, 2021-02-05:
## Bug report

**Issue details**

I ran into an issue where hitting the RTH switch caused the aircraft to rise 30m (as instructed by the `RTL_CLIMB_MIN` parameter), and then immediately descend to around 110m (with an RTL altitude set to 100m).

The odd thing is that plotting the log, I always see TAlt at the same value:

[image]

**Version**

I unfortunately do not recall the version, it was compiled from the git repo from a fairly recent commit, but not the current HEAD.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

Flying wing (FT Arrow).

**Hardware type**

An Omnibus F4 clone.

**Logs**

[log_10_2021-2-5-18-03-30.bin.zip](https://github.com/ArduPilot/ardupilot/files/5935764/log_10_2021-2-5-18-03-30.bin.zip)

rmackay9, 2021-02-06:
This should probably go in the Plane forums for investigation first before being raised as an issue.  Sorry to be strict but otherwise our to-do list turns into another support forum. https://discuss.ardupilot.org/c/arduplane/plane-4-0/153

skorokithakis, 2021-02-06:
Sure, that sounds fair, thanks.

skorokithakis, 2021-02-06:
@rmackay9 I think I know what causes this (based on more testing): `RTL_CLIMB_MIN` isn't counted as "extra altitude".

Say my RTL altitude is 70 meters, `RTL_CLIMB_MIN` is 30, and my current altitude is 120 meters. If I activate RTL the wing climbs to 150 meters (because of `RTL_CLIMB_MIN`), then *immediately descends* to 120 meters again, and starts the linear descent from 150 meters to 70 meters over the home location.

This seems like a bug to me, the craft should not waste all that energy by diving to the altitude it had before the `RTL_CLIMB_MIN` climb.

skorokithakis, 2021-02-08:
@rmackay9 @andyp1per can this be reopened? It seems like an actual bug.

peterbarker, 2021-02-09:
I've reproduced this with an autotest in #16565 .

The code is quite clear here, and the behaviour is entirely explained by it.  The question is - why would we descend straight after the climb is complete?  If the goal is to clear surrounding obstacles, perhaps we *should* maintain the higher altitude for the duration of the RTL.

skorokithakis, 2021-02-09:
Personally, I use this feature exactly for clearing obstacles. Mainly because, if there's going to be an obstacle when I initiate an RTL (manually or from a failsafe), the least likely place for an obstacle to be is in front of me.

So, I use the climb as a way to try to clear obstacles before starting to turn (which, in the worst case, would start to turn right into the obstacle that caused the failsafe).

This is itself a workaround for a feature Ardupilot lacks that other projects have, an option to climb *first* and then turn.  This is helpful because the first thing you want is to get to a safe altitude, and only then start performing maneuvers.

peterbarker, 2021-02-09:
On Mon, 8 Feb 2021, Stavros Korokithakis wrote:

> This is itself a workaround for a feature Ardupilot lacks that other projects have, an option to climb first and then turn. This is helpful because the
> first thing you want is to get to a safe altitude, and only then start performing maneuvers.

Have you met the LEVEL_ROLL_LIMIT parameter?

Note that that will affect things other than RTL, however (auto takeoff 
and landing)

skorokithakis, 2021-02-09:
Hmm, how does that help? You mean by making the turns very wide, giving the aircraft more time to climb before turning? That might help, though I find RTL_CLIMB_MIN a more effective workaround (if a bit wasteful because it climbs even if I'm above the desired altitude).

skorokithakis, 2021-03-05:
This can be closed, I have tested it and can confirm it works.

As an aside, would it be possible/easy to implement a "climb before turn" setting? I used this setting to work around that, but it is a poor substitute in many cases.

peterbarker, 2021-03-12:
Thanks @skorokithakis - the question you pose there is being attacked elsewhere.

closed 2021-03-12 by peterbarker

skorokithakis, 2021-03-12:
Yes, I implemented it in #16823, it works great.