ArduPilot/ardupilot #16533: AP_Button: BTN_FUNCx param descriptions should match RCx_OPTIONS

rmackay9, 2021-02-06:
We can improve the ease-of-use of the Buttons feature ([see wiki here](https://ardupilot.org/copter/docs/common-buttons.html)) by allowing the user to directly select the function they want attached to the button instead of having to manually look up the number on the wiki page.  This should be possible by simply copying the RCx_FUNCTION parameter descriptions @Values field to the BTN_FUNC1/2/3/4 parameter description.

4kashyadav, 2021-02-12:
Hi there i want to work on this issue please assign me this!

rmackay9, 2021-02-12:
ok!  good luck!

YashMaxy, 2021-02-24:
@rmackay9  May I work on this issue?

rmackay9, 2021-02-24:
@YashMaxy,

Sure, I haven't seen a Pr from @4kashyadav yet.  Whoever raises the PR first wins!

YashMaxy, 2021-02-24:
Thank you @rmacky9 sir.
I am new to this codebase. Can you please help me to where to start?

PR #16739 (2021-02-26): AP_Button: update BTN_FUNCx param descriptions
Just added the RCx_FUNCTION parameter description's values field to the BTN_FUNC parameter description

Resolves #16533
closed 2021-02-26 by peterbarker