ArduPilot/ardupilot #16543: Quad tailsitter mixer

galdalali, 2021-02-07:
## Bug report
I think there is a problem in quad tailsitter configuration because you can not setup the motors properly.
When I want to choose this configuration:
https://ardupilot.org/plane/docs/guide-tailsitter.html#quad-x-motor-tailsitter
I set the parameters as following:
Q_FRAME_CLASS = 10
Q_FRAME_TYPE    = 1
But it will never set quad motors mixer because of this condition:
https://github.com/ArduPilot/ardupilot/blob/ArduPlane-4.0.8/libraries/AP_Motors/AP_MotorsMatrixTS.cpp#L140
Perhaps there is a problem in documentation?

**Version**
Plane 4.0.8 stable

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad Tailsitter plane

**Hardware type**
Cube

**Logs**
No relevant logs

IamPete1, 2021-02-07:
You need `Q_FRAME_CLASS` 1 and `Q_TAILSIT_MOTMX` none 0

closed 2021-02-07 by IamPete1

galdalali, 2021-02-07:
I know that.. but I think there it's still an issue with documentation for the tailsitter matrix type

IamPete1, 2021-02-07:
Ah, sorry I thought you were reporting a code issue, this is the place for documentation issues, https://github.com/ArduPilot/ardupilot_wiki/issues or better yet https://ardupilot.org/dev/docs/common-wiki_editing_guide.html