ArduPilot/ardupilot #16548: Remove BRD_PWM_COUNT

tridge, 2021-02-07:
We should remove the BRD_PWM_COUNT parameter and instead allow any ordering. If a SERVOn_FUNCTION is not set or if a GPIO is specifically set with a *_PIN parameter then it should be available as a GPIO and disabled as PWM.
This would make it more flexible, and prevent user problems with getting BRD_PWM_COUNT right

rmackay9, 2021-02-08:
Sounds good.  if we do this we might consider extending the RCOUT banner to mention which pins are GPIO although then technically they are not RCOUTs anymore.

magicrub, 2021-02-08:
Do the PWM groupings have any impact on GPIO usage?

rain-er, 2021-06-13:
This would be quite helpful for a rover with "brushed with relay" motor and Neopixel on Matek H743.

Currently I need to reduce BRD_PWM_COUNT to 12 to free S13 for the GPIO62 for the brushed motor direction relay. But with this I also loose the possibility to connect Neopixel to the S13. And need to waste another complete timer group for Neopixel.

If I could set S1 for brushed-PWM and S2 for the direction relay-GPIO the timer group S1+S2 would be efficently used. And S13 as the only member of its timer group could still be used for Neopixel.

PR #18018 (2021-07-21): ChibiOS: remove BRD_PWM_COUNT
This removes BRD_PWM_COUNT and instead requires that SERVOn_FUNCTION be set to -1 for it to be used as a GPIO. This gives us a lot more flexibility in handling of GPIOs versus PWM outputs.
The tricky part of this will be the parameter conversion to retain existing setups

This PR also changes the default pins in several subsystems to be -1, to stop us defaulting to using a completely inappropriate pin
The PR also changes the param docs to remove the Pixhawk name, so PixhawkAUX1 becomes AUX1.

Fixes #16548
closed 2021-07-21 by tridge