ArduPilot/ardupilot #16574: Copter: Radio failsafe not working when receiver turns off

AckroydPP, 2021-02-09:
## Bug report

Hello.
A strange thing happened to me.
I have a companion computer that generates PPM signal and is connected to cube orange RC IN.
It works good, but today i noticed a problem. When i turn the companion computer off, The Cube still reads PPM (checked via logs and via mission planner also). Looks like it reads the last value when it was connected, but wont drop to zero. I wanted to set the radio failsafe, but like this, its not possible. 
On this thread i have been told that it might be a bug.
https://discuss.ardupilot.org/t/radio-failsafe-receiver-off-not-working/67357

I double checked with osciloscope that there is no PPM signal being send to cube after turning the companion computer off
Is it really a bug? or some protection feature?. Because right now when my companion computer freezes, the drone remains in flight and failsafe is not triggered.

**Version**
A.C. 4.0.6.rc1

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadcopter

**Hardware type**
Cube orange

**Logs**

rmackay9, 2021-02-09:
Let's keep discussing in the support forum and open an issue once it's confirmed to be a bug.  Sorry for the back-and-forth.

closed 2021-02-09 by rmackay9

AckroydPP, 2021-02-09:
No problem. going back there :-D