ArduPilot/ardupilot #16576: Servo(Motor) PWM Min/Max

idorutm, 2021-02-09:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REQUEST WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please do a careful search before opening this, there are already a lot of feature requests**_
If there was a previous discussion in the forum, link to it

## Feature request

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[ X ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

Dear Supporters,
I've got ver 4.2.0 of ArduSub and I'm using QGC.
I have an ESC that works between 800-2200 PWM.
I have defined my rc max/min to 2012/987 as my controller is and I have changed the Servo(Motor) Max/Min params to 2200/800 BUT in the analyze tools the servo_raw output shows me that the values are still between 1900/1100. 
I have checked in the analyze tools the rc input as well to see if there is something wrong with the values that the rc sends but they seem to be fine as I defined(2012/987).
It seems like the code overrides the servox_max/min parameters to always stay between 1900/1100.

Thanks in advance for the help!

rmackay9, 2021-02-10:
This is really a support question so I'll close this.

Normally I would suggest that the support be done on [discord](https://discuss.ardupilot.org/) but we don't have a section for Sub.  I guess much of that support happens on the BlueRobotics site although we probably should have a section here on ArduPilot as well because not all submarines use BlueRobotics hardware.

It might be that MOT_PWM_MIN and MOT_PWM_MAX are defining the range to be 1100 to 1900 but I'm not sure.

idorutm, 2021-02-10:
> This is really a support question so I'll close this.
>
> Normally I would suggest that the support be done on [discord](https://discuss.ardupilot.org/) but we don't have a section for Sub. I guess [...]

I strongly agree with you on having a sub section here since I myself don't use BlueRobotics' hardware.
I have already defined the Mot_PWM_MIN/MAX parameters to be in between 800-2200 and still the output stays on 1100-1900... If you could open a section here for that it would be so much helpful. thank you!

Williangalvani, 2023-07-13:
This is actually an issut. MOT_PWM_MIN/MAX are not being respected right now

PR #20606 (2023-07-27): Sub: tweak output function in AP_Motors6DOF
This removes the hardcoded `400` and now consumes MOT_MIN/MAX for calculating the range.
fix #16576
closed 2023-07-27 by Williangalvani