ArduPilot/ardupilot #16580: Copter/Rover/EKF3: EKF Variance after arming in Stabilize before EKF has position estimate

rmackay9, 2021-02-10:
Users may see a scary "EKF variance" message if they takeoff in stabilize mode before the EKF has a good position estimate.  We should probably modify the EKF failsafe so that it does not start until the EKF has had a good estimate.

To reproduce the issue in SITL:

- start SITL using a Copter or Rover
- wait for "EKF3 IMUx origin set" message (but don't wait for "EK3 IMUx is using GPS" message
- arm vehicle in Stabilize mode
- "EKF Variance" should be displayed almost immediately

robustini, 2021-02-10:
Problem confirmed.

Notorious811, 2021-03-16:
@rmackay9 I am arming my quadcopters in stabilize mode and then takeoff and fly them to different waypoints in guided mode, I have been haunted with EKF issues previously as mentioned here #16791 and faced about 3-4 bad flyways in last 6-7 months but thankfully this PR #16842 have solved that issue for now

Should I still backport this PR #16594 to ArduCopter v4.0.4 which I am using currently ? Will it help in avoiding any unprecedented issue related to EKF estimation innovations in future?

rmackay9, 2021-03-16:
@Notorious811,

I'm not really sure.  I'm not aware of any issues that could cause flyaways with the Copter stable release at least when using the default EKF2.  I see you've posted a similar question on one of the other PRs so perhaps another dev will have an answer.  In general we don't expect normal users to need to maintain their own safe versions, the stable release should be safe.  Maybe I'm missing something but I'm not aware of issues that mean 4.0.7 is not safe.

We've resolved this issue now in master so I will close this.

closed 2021-03-16 by rmackay9