ArduPilot/ardupilot #16587: Flow control serial port interaction

IamPete1, 2021-02-10:
There is seems to be a interaction between flow control settings on Serial1 and Serial2, in this case on a Cube Orange. Flow control was enabled on Serial1, this caused a Serial2 to become very slow. One path might be that this caused Serial2 to have the incorrect flow control setting, a alternate path could be that Serial1 was using more cpu/memory with flow control such that Serial2 could not go as fast. 

A quick scan of the flow control code and params does not show anything obvious, I have not done a deep dive, or reproduced locally. Original report from @pompecukor

WickedShell, 2021-02-10:
It's probably a slow serial device on the other side which is stopping us from being able to send data to it. The problem with this can be that if using a shared DMA stream (which 1 and 2 are shared from looking at the hwdef) this can tie up the DMA stream attempting to deliver data to the port which is being slow due to flow control indicating the other device isn't ready.

You get the same effect if you try to send to much data on a slow baud rate serial line and it's sharing DMA with other peripherals you can slow down the servicing of all the data using that DMA significantly. (I've induced slow down's in MAVLink telemetry from pushing to much data to a GPS at 115200 baud before with no flow control)

pompecukor, 2021-02-10:
> It's probably a slow serial device on the other side which is stopping us from being able to send data to it. The problem with this can be that if using a shared DMA stream (which 1 and 2 are shared [...]

As a matter of fact the device is very fast (a companion computer on serial 2) and we are running 115k baud on it (57 was not enough for the stream, for instance GPS at 25Hz, position 10Hz and so on).

Anyway it may not be entirely clear from Peters description, so I will mention it here.

To replicated the create/problem for us, we have **nothing connected** to Serial 1. Only have flow control enabled for it in the parameters.
Then we have our companion computer (without any flow control) on Serial 2 and, flow control of parametery of serial 2 is set 0, disabled.

Then the data just slows down so much that is is unusable.
To me it seems as if the Cube was waiting for CTS/RTS on serial 2 as-well. And throttling the data due to this.

tridge, 2021-02-10:
I think it will be caused by the shared DMA. Both telem1 and telem2 use DMA 2:3 with the current hwdef.dat. When we are trying to transmit on telem1 the DMA channel is locked and will remain locked until it times out. That holds up telem2 as it is waiting on the DMA channel.
I think we need to set a flag that says that the last transfer on a UART timed out if the UART is using hw flow control. Then we would do the next transfer on that UART without DMA.

tridge, 2021-02-10:
hmm, an even simpler fix would be to not start a transfer on a port with hw flow control enabled until the hw flow control pin shows that transmit is possible. That means we would never grab the DMA channel if the pin hw flow control pin constantly blocked transmit

tridge, 2021-02-10:
fix here:
https://github.com/ArduPilot/ardupilot/pull/16590

IamPete1, 2021-02-19:
Closed by https://github.com/ArduPilot/ardupilot/pull/16590

closed 2021-02-19 by IamPete1