ArduPilot/ardupilot #16589: CTUN.ASPD dropped to 0 m/s and caused throttling to 100% that leaded to oscillation and crash.

zhaodelong, 2021-02-10:
## Bug report
This has been reported to the Plane forums for investigation [.](https://discuss.ardupilot.org/t/failsafe-caused-ctun-aspd-dropping-and-throttle-increasing-to-100-and-crashed/67355)  After investigation, several users thought the synthetically calculated airspeed erroneously goes down to 0 m/s. Not sure if this a regression in 4.0.8. So try to raise as an issue here to further investigation. 

**Issue details**
The synthetic airspeed is up to first 10 minutes completely correct. After that, it erroneously goes down to 0 m/s. Assuming a low speed, the amplitudes of the rudder servos are increased, which then leads to flutter at the actual high speed (20 m/s). This leads to 100% throttle and fluttering that leaded to crash. 

_Please describe the problem_
why the synthetically calculated airspeed was suddenly so wrongly low.

**Version**
_What version was the issue encountered with_
4.0.8 Arduplane

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Flying wing: Ar Wing Pro

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Matek h743 wing

**Logs**
_Please provide a link to any relevant logs that show the issue_
Here is the video. The first 10 minutes is perfect. The issue started at 10:30.
[https://www.youtube.com/watch?v=sDYNlZNP-g0&t=630s ](url)

Here are the bin and log files. [https://drive.google.com/file/d/1eySArkrL_u3JlKsY6F5DQ3M0tsnFMukr/view?usp=sharing](url)

[image]
[image]
[image]

IamPete1, 2021-02-10:
The EKF was very upset due to Mag issues, Mission planner had a red EKF symbol and was showing ERROR COMPASS VARIANCE and BAD COMPASS HEALTH for the majority of the flight.

[image]

The EKF being confused caused the wind estimate to go bad and hence the airspeed estimate went to 0. At low airspeeds the control surface movement is amplified but because you were not at a low airspeed this pushed you into a pitch osculation.

zhaodelong, 2021-02-11:
Thanks for the root cause analysis, Peter. I close this issue now.

closed 2021-02-11 by zhaodelong