ArduPilot/ardupilot #16608: Better alternative of SD card which can support AP_Filesystem API and lua scripting

Notorious811, 2021-02-12:
## Feature request

**Is your feature request related to a problem? Please describe.**
I found issues with SD card sometimes, while reading any file from SD card, rarely i run into problems where AP_Filesystem API not able to read the file from SD card and get stuck, but after formatting the SD card, it get back to work properly again 

Another thing which i don't like that FCs which support SD cards are not much tiny or lightweight for smaller builds, i generally try to build quadcopters under 250grams weight which can also fly in auto and guided modes using GPS so using FCs like Omnibus F4 Nano V6 is more handy but they have onboard serial flash, so cannot read/write files using AP_Filesystem API.

**Describe the solution you'd like**
Better alternative for SD card for reading/writing files using AP_filesystem api or with lua scripting, i have seen people recommend using serial flash memories for more reliable and faster performance compared to SD cards. So supporting AP_filesystem api for read/write operations and supporting lua scripting from onboard serial flash will be really nice feature.

**Platform**
[ * ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

IamPete1, 2021-02-12:
you can include scripts in ROMFS by placing them in a directory `scripts` of the hwdef folder for the flight controller your building for.

Notorious811, 2021-02-12:
> you can include scripts in ROMFS by placing them in a directory `scripts` of the hwdef folder for the flight controller your building for.

I know about that but I think its just meant for leftover internal flash memory of STM2 MCU after flashing the firmware, isn't it ? I am looking if same can be implemented for winbond or micron external SPI flash which can run both lua script and also support AP_filesystem API

ingeer314, 2024-08-08:
Any news regarding this topic?

andyp1per, 2024-08-08:
Manufacturers have started to ship sdcard flash chips - I think that will make this problem go away.