ArduPilot/ardupilot #16614: AP_Arming/EKF3: add detail to "Need Position Estimate" pre-arm failure

rmackay9, 2021-02-13:
4.1 may display the "Need Position Estimate" message when the EKF does not yet have a good position estimate.  If possible we should display what the issue is.  e.g. GPS, IMU or something else.

I think if the GPS doesn't have enough satellites we display a message.  If it's the IMU I'm sure we don't print any extra information.

hi6otb, 2022-09-22:
I'm unable to arm due to this error.
Can I still use the following features in version 4.2.3?
I need to use ROS and T265.

https://ardupilot.org/dev/docs/ros-vio-tracking-camera.html

Dkyouma, 2024-11-28:
Is there anything to be done? I have the same problem but in different way i have a raspberry pi script that command my pixhawk but i have a statement 
Prearm : need a postion estimate 
And it is not arming any good solution i have tried to disable my gps but nothing works

Dkyouma, 2024-11-28:
> Hi @Dkyouma <https://github.com/Dkyouma>,
>
> Our pre-arm error message wiki page is here
> <https://ardupilot.org/copter/docs/common-prearm-safety-checks.html>.
>
> If the vehicle is being armed in a flight mode that requires a position
> estimate then it needs a GPS with a good view of the sky. If a GPS isn't
> possible then maybe have a look at our non-GPS wiki page which is here
> <https://ardupilot.org/copter/docs/common-non-gps-navigation-landing-page.html>.
> Alternatively consider trying to arm in a mode that doesn't require a
> position estimate

Can you be more specific, what you mean is the my pixhawk model only
supports gps so i have to buy another model to apply non gps navigation?

rmackay9, 2024-11-29:
Hi @Dkyouma,

These questions really belong on [the support forums](https://discuss.ardupilot.org/c/arducopter/copter-45/174) but the short answer is that it doesn't depend upon the flight controller.  It depends upon the sensors connected to the flight controller.  Please have a peek at the [Non-GPS page](https://ardupilot.org/copter/docs/common-non-gps-navigation-landing-page.html) if you're doing indoor flight

Dkyouma, 2024-11-29:
So i have buy some sensors to do that mission it is not about the mission planner settings?

rmackay9, 2024-11-29:
@Dkyouma,

I guess you're asking the age old question, "why does my vehicle need a GPS?  Why can't it just estimate it's position using the IMU?".  the answer is that the IMUs in modern flight controllers are not accurate enough to do this.  They all have noise and drift which when integrated (which is what is required to convert an acceleration into a velocity and then position) results in the position estimate diverging from reality in a matter of seconds.