ArduPilot/ardupilot #16631: AP_OpticalFlow/EKF: add inflight optical flow calibration

rmackay9, 2021-02-16:
Optical Flow can allow good non-GPS position estimation if the user accurately [calibrates the sensor](https://ardupilot.org/copter/docs/common-optical-flow-sensor-setup.html).  The calibration procedure though is difficult to get right though so we should add support for an inflight calibration procedure.  This would involve the user attaching both a GPS and an optical flow and then the EKF should be able to estimate the expected flow measurements, compare with the actual flow measurements and then calculate the best scaling values.

I have made a bit of progress in this branch but it is definitely not complete: https://github.com/rmackay9/rmackay9-ardupilot/commits/optflow-calibration

Shubhupatel, 2021-03-05:
Hello, Dear @rmackay9 Interested to work on this issue for GSoC.

Shubhupatel, 2021-03-14:
Hello @rmackay9 Please tell me about the project.

rmackay9, 2022-02-21:
This has been completed now so closing

closed 2022-02-21 by rmackay9