ArduPilot/ardupilot #16632: AC_Avoid: add support for MNT EYE depth cameras

rmackay9, 2021-02-16:
ArduPilot already supports object avoidance using the Intel RealSense 435 and 455 depth cameras.  We should also add support for the [MYNT EYE depth cameras](https://www.mynteye.com/pages/products).

This project involves:

- writing python scripts to pull the data from the depth camera and package them into OBSTACLE_DISTANCE and/or OBSTACLE_DISTANCE_3D mavlink messages
- producing APSync images for at least one companion computer (RPI4?) that can run the above script

rishabsingh3003, 2021-02-16:
The best way to achieve this is to probably use all the Mavlink bits from the script linked in the issue, replace the D4xx connection and filtering code with MNT EYE depth cameras. 
Basically the only change from the script is to retrieve the depth matrix using MNT EYE's API instead of Intel RealSense python API.

IamPete1, 2022-09-10:
@rmackay9 @rishabsingh3003 Any update?

rishabsingh3003, 2022-09-11:
@IamPete1 I never got hold of the MNT EYE cameras