ArduPilot/ardupilot #16633: AP_Proximity: Driver bug with SF40c

Hasi123, 2021-02-16:
## Bug report

**Issue details**

There is a bug with the SF40c´s driver, which produces sharp spikes in the distance readings and in reaction causes the copter to suddenly pitch or roll. The issue has been discovered by multiple users and discussed on the forum [here ](https://discuss.ardupilot.org/t/errorous-readings-with-sf40c/61852)and [here](https://discuss.ardupilot.org/t/copter-4-unstability-with-sf40-lidar/63588).

I have even sent in the unit to LightWare and they determined, that it was functioning properly and there is a problem with ArduCopter´s drivers.
> We have been testing and retesting your unit since the units been reset and we cannot fault the unit. 
> We have tested the unit using Pixhawk and from what we can see, the unit still reports the same quick spike. This leads me to believe that this may be the way that the Ardupilot stack is interpreting the distance data. 

**Version**
Copter 3.6 and 4.0

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Reproduced on Quad and Hexa

**Hardware type**
Reproduced on CubeBlack and Pixhawk 4

**Logs**

[SF40c issue.zip](https://github.com/ArduPilot/ardupilot/files/5987492/SF40c.issue.zip)

rishabsingh3003, 2021-02-16:
@Hasi123 , we are in the process of adding low pass filters, and limiting avoidance acceleration as part of Simple Avoidance feature. 
What this will do is:
1. Sudden sharp noises, that are visible only for a short period of time, will most probably be eliminated (as is your case)
2. Even if the noise does get captured by the driver, the second bit will make sure that the reaction is smooth and not very violent.

After https://github.com/ArduPilot/ardupilot/pull/16604 is merged in, will it possible for you to test if everything is working? It would help us a lot in making avoidance experience better.

Hasi123, 2021-02-16:
I am ready and eager to test any improvements that you can offer.

Just for discussion: While being a great addition to the code, unfortunately this only tries to circumvent the problem. If you limit the acceleration too much, the copter wouldn´t be able to stop in time when the vehicle is approaching at high speeds.
Would this also effect the Bendy Ruler method in AUTO flight modes?

rishabsingh3003, 2021-02-16:
great, thanks! I will ping you when the PR is ready.
How does the filter circumvent the issue? I am assuming that the log that you have provided is sensor noise which most sensors have. The filter will help fix this.

The acceleration feature comes with a parameter. So you can decide how much you want to limit the acceleration (or not to limit it at all)

This will not  effect BendyRuler as the path planning algo's use WP navigation. So any WP param's will affect BendyRuler.

rmackay9, 2021-02-16:
The issue's description is not at all accurate so I will update it in a moment.  The sensor itself is producing the glitches, not the AP driver.  On the AP side we can add a filter to try and remove the glitches but the underlying issue is with the sensor itself.

Hasi123, 2021-02-16:
> 
> 
> I don't think the issue's description is accurate so I will update it in a moment. As far as we know, the sensor itself is producing the glitches, not the AP driver. On the AP side we can add a filter to try and remove the glitches but I believe the underlying issue is with the sensor itself.

Ironically LightWare told me the exact opposite according to their tests.

Hasi123, 2021-02-16:
> 
> How does the filter circumvent the issue? I am assuming that the log that you have provided is sensor noise which most sensors have. The filter will help fix this.

The problem is not normal noise from the sensor, but the sudden spikes down to a specific value, which is always the same. In my case it is always 4.33

rmackay9, 2021-02-16:
@Hasi123,

Yes, indeed the 4.33 is very suspicious.  I have two thoughts at this point:

1. The AP driver asks the sensor for its motor state at 1hz and maybe this is upsetting the sensor somehow and it glitches.  I suspect when lightware tests stand-alone they don't have this extra query being sent to the sensor so this would explain the discrepancy
2. We may have a subtle bug somewhere in the driver.  I don't immediately see any issues but we should review the code vs the spec to see if there are any mistakes.

Hasi123, 2021-02-18:
@rishabsingh3003 i flight tested your PR and everything works fine.
However I couldn´t observe the spikes even in the raw logs, but the filtering works fine. Just for gigs I unplugged the RX line of the SF40 on the second flight, but since no spikes were observed in the first flight, there is no additional information from this.

Logs: [Flight_PRX_low_pass_filter.zip](https://github.com/ArduPilot/ardupilot/files/6002734/Flight_PRX_low_pass_filter.zip)

Were there any changes in master since 4.0.4 that would fix the spikes?

IamPete1, 2021-09-12:
Sounds like this has been fixed, please re-open if you can reproduce.

closed 2021-09-12 by IamPete1