ArduPilot/ardupilot #16636: Flashing IOMCU does not seem to work as expected (tested with Holybro Durandal)

rotorman, 2021-02-16:
## Bug report
I created a custom AP_IOMCU firmware (iofirmware_lowpolh.bin) to fix an analog RSSI scaling issue on Holybro Durandal (see PR #16607). Unfortunately flashing it to IO µC was more tedious than it should have been due to likely problem in AP_IOMCU::upload_fw().

The I/O µC fw is typically embedded into AP firmware, and by holding the safety button on power-up should trigger flashing the I/O µC (STM32F100C8 on Holybro Durandal). Unfortunately AP is not able to flash the I/O µC as I have found.

With PX4 fw, this is very unproblematic. So my current solution was to flash first PX4 fw to Durandal, use QGC _nsh shell_, where I was able to flash the self compiled ArduPlane fw file iofirmware_lowpolh.bin to I/O MCU by issuing:
`px4io forceupdate 14662 /fs/microsd/iofirmware_lowpolh.bin`
(14662 is the PX4IO_REBOOT_BL_MAGIC)

**Version**
ArduPlane 4.0.8

**Platform**
I tested only with:
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
WING

**Hardware type**
Holybro Durandal with STM32H743 + STM32F100C8

auturgy, 2021-02-17:
Can you please state the steps to reproduce?

rotorman, 2021-02-17:
Yep, I am aware of that. What I used, was the python script _Tools/scripts/build_iofirmware.py_ 
This builds the iofirmware for high and low polarity heater configurations and copies them both into _Tools/IO_Firmware_.

The steps I used, thus are:

1. clone AP repo
2. Modify two source files as in my #16607
3. run _Tools/scripts/build_iofirmware.py_ 
4. run _./waf configure --board Durandal_
5. run ./waf plane
6. using Mission Planner, flash build/Durandal/bin/arduplane.apj to Durandal
7. power off Durandal
8. hold safety switch and keep holding it and power on Durandal
9. but I/O MCU does not get flashed

Similar procedure with PX4 fw (flashed with QGC), flashes the IO_MCU (can be seen via different LED blinking of I/O MCU), but naturally the in PX4 fw embedded I/O MCU fw, not the one I compiled. Thus resorted to copying the _iofirmware_lowpolh.bin_ to microsd card and flashing it via _px4io forceupdate_ from nsh. This worked flawlessly and also the new functionality from #16607 is there.

*****

I also tried flashing the custom built arduplane_bl.hex with STM32CubeProgrammer. Also explicitly erased the chip (H743 of Durandal, no access to I/O MCU from STM32CubeProgrammer via Durandal USB) - this did not make a change.

rotorman, 2021-02-18:
@auturgy can you relate to the steps I listed?

auturgy, 2021-02-18:
I’ve pulled your change into a branch and built it. Will be tonight or tomorrow before I can test: need to dig out a H7+iomcu board.

rotorman, 2021-02-19:
Awesome, thanks! For easier analog RSSI testing I added a pot to Durandal (getting 3.3V power from neighboring DSM port):

[image]

auturgy, 2021-02-21:
Not replicated with my test (acknowledging I tested on CubeOrange not Durandal).  On today's Master with your iomcu change, and the new iomcu bin committed to the branch, the board updated iomcu fw automatically when flashed:
APM: Initialising ArduPilot
IOMCU: update complete
IOMCU startup
Curious.

rotorman, 2021-02-21:
I cannot say about CubeOrange, do not have it to test with.

Just curious - does the CubeOrange have the same analog RSSI problem at all? Maybe Holybro Durandal electrical circuit differs from "std." IOMCU? Holybro uses for analog RSSI direct connection to STM32F100 pin PA5, meaning the range is 0 to 3.3V. In original ArduPilot code there was a multiplication factor 9900, which is OKish for VSERVO (as this has voltage divider in front), but not for analog RSSI. 
Anyhow, I fixed this and had no luck flashing the modified iofirmware_lowpolh.bin embedded in ArduPlane fw. I tried the tips I could find in docs [here](https://ardupilot.org/copter/docs/common-safety-switch-pixhawk.html#using-the-safety-switch-to-force-i-o-board-firmware-update) and [here](https://ardupilot.org/dev/docs/pixhawk-force-px4io-flash.html), did not work. First, when I manually flashed it with nsh using PX4 firmware as intermediate step, this worked.

rotorman, 2021-02-24:
Hi @auturgy I just flashed v4.0.9 release to my Durandal, and indeed AP_IOMCU flashing must work, as now the RSSI scaling is saturated again above 41% of 3.3V range roughly. So I will close this issue and hope that someone takes the PR in #16607 more seriously.
Thank you for looking into this!

closed 2021-02-24 by rotorman