ArduPilot/ardupilot #16643: Rcx_option to ArduSub

barpeleds, 2021-02-17:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REQUEST WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please do a careful search before opening this, there are already a lot of feature requests**_
If there was a previous discussion in the forum, link to it

## Feature request

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[X] Submarine

**Additional context**
As been discussed on Discord, I've tackled an issue finding a rc "motor emergency stop" for ArduSub.
In other platforms its possible to set in rcx_option the value of (31) for motor emergency stop.

Thanks for your assistance!

ES-Alexander, 2025-12-05:
I believe this was enabled by #29742.
It is at least the case that [the parameters are detected as available](https://ardupilot.org/sub/docs/parameters.html#rc1-option-rc-input-option) (with the requested option).

@Hwurzburg do you know whether your RC inclusion adds general support for ArduPilot's RC options?

Hwurzburg, 2025-12-05:
> I believe this was enabled by [#29742](https://github.com/ArduPilot/ardupilot/pull/29742). It is at least the case that [the parameters are detected as [...]

it should enable all vehicle independent options when RC is included and enabled

closed 2025-12-05 by ES-Alexander